mavlink/MAVSDK #2023: ArduCopter – fly_mission.cpp example

ygratz, 2023-04-18:
Hi,
I tried running the fly_mission.cpp when connecting to SITL – ArduCopter.
I get the following indications:
  Uploading mission…
  Falling back to gimbal protocol v1 (mission_impl.cpp:128)
  Invalid sequence for ArduPilot items (mavlink_mission_transfer.cpp:249)
  Mission upload failed: Error, exiting.

Do you know why is that?
After reading some of the forums – I think this might be because for ArduCopter, I must set the home_location as the first mission_item. Is that the reason?
And if so – what would be a correct way to implement this in C++?
Thanks.

JonasVautherin, 2023-04-18:
> Is that the reason?

I am not entirely sure of the root cause, but the error you see is triggered here: https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/mavlink_mission_transfer.cpp#L249. I don't know if this ever worked (I know that the `mission_raw` should work, but I don't think the `mission` plugin ever did).

Would you like to debug this?

julianoes, 2023-04-18:
> However, maybe the mission plugin is not sending a valid list of items for Ardupilot.

I think that's right.

ykhedar, 2023-04-22:
Hi @ygratz, for reference on how to create a valid mission for Ardupilot, please have a look at the integration test:

https://github.com/mavlink/MAVSDK/blob/2caa1c57c8ab2cfaf82fe4c976c991f0c691f28d/src/integration_tests/mission_raw_import_and_fly.cpp#L90

and

https://github.com/mavlink/MAVSDK/blob/2caa1c57c8ab2cfaf82fe4c976c991f0c691f28d/src/integration_tests/mission_raw_import_and_fly.cpp#L151

OliverHeilmann, 2023-05-30:
I just posted in #728 with the same issue. Perhaps I have shed some light on the problem and the expected ArduPilot format now? 

I definitely could be wrong, but I think the ```MissionRaw::MissionItem``` does not contain the correct arguments...

julianoes, 2023-05-31:
I was wrong. The sequence just increases normally from 0, even if 0 is the home position.

closed 2023-05-31 by julianoes