mavlink/MAVSDK #2029: Mission upload failed: Unsupported, exiting.

ShanbinZhang, 2023-05-05:
When running fly_missions in the example, an error message is displayed, stating Mission upload failed: Unsupported, exiting. In enum class Result: Unsupported, /**&lt; @brief Mission downloaded from the system is not supported. */    But, I don't understand what is: the Mission downloaded from the system is not supported.   

Thank you!

JonasVautherin, 2023-05-05:
Are you using Ardupilot?

ShanbinZhang, 2023-05-05:
> Are you using Ardupilot?

Yes, is Ardupilot , and the virtual environment is built on the development board of NVIDIA NX. In addition, I am using MAVSDK version 0.44. Because only this version can be compiled in NVIDIA NX.

The compilation steps are as follows:
1. git clone --branch v0.44.0 https://github.com/mavlink/MAVSDK.git
2. git submodule update --init --recursive
3. cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
4. cmake --build build/default

JonasVautherin, 2023-05-05:
The mission plugin does not support Ardupilot yet (it would require some contributions :innocent:). For now you have to use the mission_raw plugin, which works :+1:

ShanbinZhang, 2023-05-06:
> 
Thank you. Does that mean I can not use fly_missions in the example? I found that fly_qgc_mission have to use the mission_raw plugin. But I still didn't succeed, (./fly_qgc_mission udp://:55408)

JonasVautherin, 2023-05-06:
Not sure, I guess it depends on the elements you have in your mission?

ShanbinZhang, 2023-05-11:
> Not sure, I guess it depends on the elements you have in your mission?
Thank you. I have been tested successfully

JonasVautherin, 2023-05-11:
Can you maybe list the elements that were unsupported?

Also can this issue be closed now?

ShanbinZhang, 2023-05-15:
I used fly_ qgc_ Mission, the command I initiated earlier was incorrect. This issue can be closed now, thank you!

closed 2023-05-15 by JonasVautherin

xiugou798, 2024-03-14:
> I used fly_ qgc_ Mission, the command I initiated earlier was incorrect. This issue can be closed now, thank you!

I also encountered the same problem. Can you share how you use fly_qgc_mission?

Thank you!