mavlink/MAVSDK #2031: Changing mode to GUIDED

ygratz, 2023-05-11:
Hi,
I ran the takeoff_and_land example.
In some cases it did not takeoff. Now, I figured out that this was because the arducopter wasn't in GUIDED mode.
What can be the easiest way to command it to GUIDED mode ? 
At what class (Action, System,  ?) is that implemented.
Thanks

ygratz, 2023-05-11:
libmavsdk-dev_1.4.6_ubuntu20.04_amd64.deb

JonasVautherin, 2023-05-11:
Oh, it is fixed in upstream (which will become MAVSDK v2 at some point), but not in the v1.4.x branch: https://github.com/mavlink/MAVSDK/blob/v1.4.13/src/mavsdk/plugins/action/action_impl.cpp.

So you have two solutions:
1. Build MAVSDK from source from the main branch, so that you get the fix
2. Open a PR backporting the fix from main to the v1.4 branch, and then we could make a v1.4.14 release

I don't know how many similar fixes are in `main` and not in `v1.4` (apparently `v1.4` is currently 438 commits behind `main`: https://github.com/mavlink/MAVSDK/tree/v1.4). If I was using Ardupilot, I would personally build MAVSDK from source to make sure I don't miss other similar fixes.

But that's your choice :blush:.

ykhedar, 2023-06-01:
I would suggest to use the main branch for building the library as well since there are many Ardupilot related changes which are not in the latest release. Closing for now, please reopen if the issue persists with the main branch as well.

closed 2023-06-01 by ykhedar

OliverHeilmann, 2023-09-23:
Hello,

I know this issue is closed but I'd like to add to it and perhaps suggest that it is reopened... I have changed to main branch, built from source etc. and the `takeoff` MavSDK action does work now... sometimes.

I have been using the ArduPilot SITL environment to test my code and have noticed that I sometimes get a seg fault when using the `takeoff` command. I have tried out quite a few things and I *think* I have boiled it down to the `takeoff` command switching to `GUIDED` mode when there is no `position` as outlined in the logs shown in the image below.

[image]

Based on [action.h](https://github.com/mavlink/MAVSDK/blob/48f1ad0325ecdcb8075d7ffd0cd7a8323af12fb3/src/mavsdk/plugins/action/include/plugins/action/action.h#L84), I am getting a takeoff response of 13 which relates to `Failed` as shown below:

```cpp
    enum class Result {
        Unknown, /**< @brief Unknown result. */
        Success, /**< @brief Request was successful. */
        NoSystem, /**< @brief No system is connected. */
        ConnectionError, /**< @brief Connection error. */
        Busy, /**< @brief Vehicle is busy. */
        CommandDenied, /**< @brief Command refused by vehicle. */
        CommandDeniedLandedStateUnknown, /**< @brief Command refused because landed state is
                                            unknown. */
        CommandDeniedNotLanded, /**< @brief Command refused because vehicle not landed. */
        Timeout, /**< @brief Request timed out. */
        VtolTransitionSupportUnknown, /**< @brief Hybrid/VTOL transition support is unknown. */
        NoVtolTransitionSupport, /**< @brief Vehicle does not support hybrid/VTOL transitions. */
        ParameterError, /**< @brief Error getting or setting parameter. */
        Unsupported, /**< @brief Action not supported. */
        Failed, /**< @brief Action failed. */
    };
```

I have also noticed that if I let the simulator "settle" for some time, I don't get this seg fault, presumably because there is a position. I have tested this on a real drone as well and have a similar issue, though I cannot simply let the drone settle as real GPS dropout causes a loss of position at any arbitrary time.

In any case, I am unable to catch this error so it pretty much breaks my code, though I also need it for a drone takeoff.

Thanks for your help!

ykhedar, 2023-09-25:
Hi Oliver, thanks for reporting the issue. Could you maybe try by changing the takeoff code in your local MAVSDK to switch to Stabilize flight mode instead of guided mode before takeoff? Please check it on SITL ;) Relevant code is here.

https://github.com/mavlink/MAVSDK/blob/48f1ad0325ecdcb8075d7ffd0cd7a8323af12fb3/src/mavsdk/plugins/action/action_impl.cpp#L387

Stabilize mode does not need position so should not create this particular issue:
https://ardupilot.org/copter/docs/stabilize-mode.html#stabilize-mode

What is your use-case? You would need position fix anyways before carrying out an auto mission right? In such a case waiting for a good fix before activating takeoff should be prudent. Or are you flying indoors?

Arducopter allows to takeoff in a selected number of flight modes out of which Guided was selected since MAVSDK is supposed to control the drone from such a mode. But I am open to discussing this further to find the best suitable mode for takeoff for most scenarios. Would also discuss this with the ArduPilot Dev team shortly.

OliverHeilmann, 2023-09-27:
Hello, thanks for your response. Prior to your message, I messed around with the `takeoff_async` and noticed that it can still fail/ throw errors but at least it doesn't throw a seg-fault (so you can handle the cases where this happens):

This code snippet is a little weird in implementation because I am turning an async method into a blocking one but this works and gets the "job done" so to speak. This one doesn't throw errors because everything stays in scope until the takeoff result is returned, thread notified etc. (if you use it perhaps put a timeout or something....).
```cpp
void MyClass::takeoffSync(mavsdk::Action::Result& result)
{
    LOG_INFO("Mavlink::takeoffSync() Method called!");
    std::condition_variable cv;
    std::mutex mtx;
    std::unique_lock<std::mutex> lock(mtx);
    bool done = false;

    // Takeoff is an async function, so we need to wait for it to complete before returning
    action->takeoff_async([&](mavsdk::Action::Result tempResult) { 
        result = tempResult;
        done = true;
        cv.notify_one();
    });
    
    // Wait until the async function has completed...
    cv.wait(lock, [&]{ return done; });
}
```

This one follows the `mavsdk` structure a little more closely with the `future` and `promise` approach. This one will throw errors as you'd expect because of the scope issue mentioned above. You can either do a timer or catch the error now without the seg-fault issue.
```cpp
std::future<mavsdk::Action::Result> MyClass::takeoffAsync()
{
    LOG_INFO("Mavlink::takeoffAsync() Method called!");

    auto sharedPromise = std::make_shared<std::promise<mavsdk::Action::Result>>();
    auto future = sharedPromise->get_future();

    std::atomic<bool> isValueSet{ false };
    action->takeoff_async([sharedPromise, &isValueSet](mavsdk::Action::Result result) {
        if (!isValueSet.exchange(true)) {
            sharedPromise->set_value(result);
        }
    });

   // perhaps add some timer here...

    return future;
}
```

Anyway, @ykhedar I am just confirming that the async approach works for a drone in simulation and for real. I *think* (will check again) that the method still changes it to `GUIDED` though... I am going to try the `takeoff_async_apm` next to see the performance.

In any case, I think the blocking `takeoff` could be fixed with one of my two proposed solutions? Would be good to avoid creating seg-faults in general.

One issue I have with the `STABILIZED` mode in simulation is that the drone will fall out of the sky during the moments when in that mode. This is because it expects a throttle input from somewhere. If there is none, the system drops. See the [flight mode documentation](https://ardupilot.org/plane/docs/flight-modes.html) for what I mean- anything with two `A`'s (e.g. `GUIDED`) means the sim will keep the drone level. Perhaps include this in your dev-team discussions...

Hope this was useful for you and thank you for your help!