mavlink/MAVSDK #2037: mission_raw   missionitem

ShanbinZhang, 2023-05-18:
{
    MissionRaw::MissionItem new_raw_item_nav{};
    new_raw_item_nav.seq = _seq;
    new_raw_item_nav.frame = _frame; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav.command = _command; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav.current = _current;
    new_raw_item_nav.autocontinue = _autocontinue;
    new_raw_item_nav.param1 = _param1; // Hold
    new_raw_item_nav.param2 = _param2; // Accept Radius
    new_raw_item_nav.param3 = _param3; // Pass Radius
    new_raw_item_nav.param4 = _param4; // Yaw
    new_raw_item_nav.x = int32_t(std::round(_x * 1e7));
    new_raw_item_nav.y = int32_t(std::round(_y * 1e7));
    new_raw_item_nav.z = float(_z);
    new_raw_item_nav.mission_type = _mission_type;
    return new_raw_item_nav;
}
For the above structures, does the longitude and latitude have to be multiplied by 10 to the seventh power? 2. Must the first waypoint be set for takeoff? 3. Can we take off first and then execute the route? 4. What does current mean? 5. Can Param 1-4 be set to 0.0f in all routes? I built my own missionitem, but drones always fly in one direction and are not executing a course

ShanbinZhang, 2023-05-18:
std::vector<MissionRaw::MissionItem> reference_items;
        MissionRaw::MissionItem new_raw_item_nav;
        new_raw_item_nav.seq = 0;
        new_raw_item_nav.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        new_raw_item_nav.command = MAV_CMD_NAV_TAKEOFF; // MAV_CMD_NAV_WAYPOINT
        new_raw_item_nav.current = 1;
        new_raw_item_nav.autocontinue = 1;
        new_raw_item_nav.param1 = 0.0f; // Hold
        new_raw_item_nav.param2 = 0.0f; // Accept Radius
        new_raw_item_nav.param3 = 0.0f; // Pass Radius
        new_raw_item_nav.param4 = 0.0f; // Yaw
        new_raw_item_nav.x = -35.36326310;
        new_raw_item_nav.y = 149.16523800;
        new_raw_item_nav.z = 50.0f;
        new_raw_item_nav.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        MissionRaw::MissionItem new_raw_item_nav1;
        new_raw_item_nav1.seq = 1;
        new_raw_item_nav1.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        new_raw_item_nav1.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
        new_raw_item_nav1.current = 0;
        new_raw_item_nav1.autocontinue = 1;
        new_raw_item_nav1.param1 = 0.0f; // Hold
        new_raw_item_nav1.param2 = 0.0f; // Accept Radius
        new_raw_item_nav1.param3 = 0.0f; // Pass Radius
        new_raw_item_nav1.param4 = 0.0f; // Yaw
        new_raw_item_nav1.x = -35.36326120;
        new_raw_item_nav1.y = 149.16524010;
        new_raw_item_nav1.z = 50.0f;
        new_raw_item_nav1.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        // MissionRaw::MissionItem new_raw_item_nav2;
        // new_raw_item_nav2.seq = 2;
        // new_raw_item_nav2.frame = MAV_FRAME_MISSION; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        // new_raw_item_nav2.command = MAV_CMD_IMAGE_START_CAPTURE; // MAV_CMD_NAV_WAYPOINT
        // new_raw_item_nav2.current = 0;
        // new_raw_item_nav2.autocontinue = 1;
        // new_raw_item_nav2.param1 = 0.0f; // Hold
        // new_raw_item_nav2.param2 = 0.0f; // Accept Radius
        // new_raw_item_nav2.param3 = 1.0f; // Pass Radius
        // new_raw_item_nav2.param4 = 0.0f; // Yaw
        // new_raw_item_nav2.x = 0;
        // new_raw_item_nav2.y = 0;
        // new_raw_item_nav2.z = 0.0f;
        // new_raw_item_nav2.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        MissionRaw::MissionItem new_raw_item_nav3;
        new_raw_item_nav3.seq = 2;
        new_raw_item_nav3.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        new_raw_item_nav3.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
        new_raw_item_nav3.current = 0;
        new_raw_item_nav3.autocontinue = 1;
        new_raw_item_nav3.param1 = 0.0f; // Hold
        new_raw_item_nav3.param2 = 0.0f; // Accept Radius
        new_raw_item_nav3.param3 = 0.0f; // Pass Radius
        new_raw_item_nav3.param4 = 0.0f; // Yaw
        new_raw_item_nav3.x = -35.36325940;
        new_raw_item_nav3.y = 149.16523880;
        new_raw_item_nav3.z = 50.0f;
        new_raw_item_nav3.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        MissionRaw::MissionItem new_raw_item_nav4;
        new_raw_item_nav4.seq = 3;
        new_raw_item_nav4.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        new_raw_item_nav4.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
        new_raw_item_nav4.current = 0;
        new_raw_item_nav4.autocontinue = 1;
        new_raw_item_nav4.param1 = 0.0f; // Hold
        new_raw_item_nav4.param2 = 0.0f; // Accept Radius
        new_raw_item_nav4.param3 = 0.0f; // Pass Radius
        new_raw_item_nav4.param4 = 0.0f; // Yaw
        new_raw_item_nav4.x = -35.36325950;
        new_raw_item_nav4.y = 149.16523720;
        new_raw_item_nav4.z = 50.0f;
        new_raw_item_nav4.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        MissionRaw::MissionItem new_raw_item_nav5;
        new_raw_item_nav5.seq = 4;
        new_raw_item_nav5.frame = MAV_FRAME_MISSION; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
        new_raw_item_nav5.command = MAV_CMD_NAV_RETURN_TO_LAUNCH; // MAV_CMD_NAV_WAYPOINT
        new_raw_item_nav5.current = 0;
        new_raw_item_nav5.autocontinue = 1;
        new_raw_item_nav5.param1 = 0.0f; // Hold
        new_raw_item_nav5.param2 = 0.0f; // Accept Radius
        new_raw_item_nav5.param3 = 0.0f; // Pass Radius
        new_raw_item_nav5.param4 = 0.0F; // Yaw
        new_raw_item_nav5.x = 0;
        new_raw_item_nav5.y = 0;
        new_raw_item_nav5.z = 0.0f;
        new_raw_item_nav5.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

        reference_items.push_back(new_raw_item_nav);
        reference_items.push_back(new_raw_item_nav1);
        //reference_items.push_back(new_raw_item_nav2);
        reference_items.push_back(new_raw_item_nav3);
        reference_items.push_back(new_raw_item_nav4);
        reference_items.push_back(new_raw_item_nav5);

I built the above, but I don't know where the problem is

julianoes, 2023-05-18:
What's the error? Does it upload but not do the right thing? Can you debug it in SITL?

ShanbinZhang, 2023-05-18:
> What's the error? Does it upload but not do the right thing? Can you debug it in SITL?

The drone did not fly according to the designated GPS point

julianoes, 2023-05-18:
Does it fly somewhere else? Can you download the mission items using QGC to see if they are correct on the map?

ShanbinZhang, 2023-05-19:
> Does it fly somewhere else? Can you download the mission items using QGC to see if they are correct on the map?
Thank you. I have found the problem. The longitude and latitude need to be *10e7. I have another question, will mavsdk automatically convert 353632612 to 35.3632612?

closed 2024-05-27 by julianoes