mavlink/MAVSDK #2039: MissionRaw::Result MissionRaw::pause_mission()

ShanbinZhang, 2023-05-19:
During the course of airline flight, when pause_mission , the UAV switches to ALTHOLD mode. But why does the UAV automatically land until it lands

julianoes, 2023-05-20:
It should go into Hold mode, not altitude hold, at least for PX4. Are you using PX4 or ArduPilot?

ShanbinZhang, 2023-05-24:
using ArduPilot

JonasVautherin, 2023-05-24:
Wait... shouldn't [this](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L903) read:

```
static_cast<float>(flight_mode_to_ardupilot_copter_mode(new_mode));
```

Right now, `new_mode` is not used, I think :thinking:.

Also there is a difference between [v1.4.x](https://github.com/mavlink/MAVSDK/blob/v1.4.13/src/mavsdk/core/system_impl.cpp#L998) and [main](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/system_impl.cpp#L955), where the former would go (without the bug above) to `ardupilot::CopterMode::AltHold;` and the latter to `ardupilot::CopterMode::Loiter;`.

So it would be worth trying on the `main` branch, I would say.

JonasVautherin, 2023-05-24:
@ShanbinZhang: do you think you could test my PR above?

closed 2024-05-27 by julianoes