mavlink/MAVSDK #2041: Including the sdk via FetchContent to the project errors on build

tlapik123, 2023-05-19:
When I try to include the library to my project using FetchContent it errors when building.

The error: 
```
[build] <path>/build/_deps/mavsdk-build/src/mavsdk/core/include/mavsdk/mavlink_include.h:3:10:  fatal error: mavlink/v2.0/common/mavlink.h: No such file or directory
[build]     3 | #include "mavlink/v2.0/common/mavlink.h"
[build]       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] compilation terminated.
```
I looked into the code and it seems that only that the install target would work correctly, but I don't want to install I just want to build it and use it :sweat_smile:  

Is there something that I'm doing wrong? Is there some option to set?
Thank you for your time.

julianoes, 2023-05-20:
This will soon change actually: https://github.com/mavlink/MAVSDK/pull/2028

I suggest to use that branch already.

Alternatively, are you aware of this helper script? https://github.com/mavlink/MAVSDK/blob/main/tools/generate_mavlink_headers.sh

tlapik123, 2023-05-22:
Thank you very much. I wasn't aware of those.
But as I can see you merged the change 2d ago so I don't even have to worry about the branch nor script now.
That is great news :)

closed 2023-05-22 by tlapik123