mavlink/MAVSDK #2043: vscode - debug: step into MAVSDK functions

ygratz, 2023-05-21:
Hi,
I'm using VSCode to build and debug takeoff_and_land example.
I opened VSCode for the MAVSDK (>MAVSDK\code .) and build the library using CMake.
And then opened another VSCode for the takeoff_and_land example build (using CMake) and debug it.
It works ok. I open SITL and I manage to connect to the copter and see the cmds work.
However - I do not manage to step-into any of the MAVSDK library functions.
For example: action.takeoff(); it doesn't step-into the function, rather it passes it.

What shall I do in order to be able to step-into the library functions?
Thanks

julianoes, 2023-05-21:
I don't know VSCode well but one thing to check is that you're using the debug build, and not release.

julianoes, 2023-05-22:
Ok, but is the library also debug build? Like did you build mavsdk in debug, or are you using the library installed from the deb packet? That one would be release.

ygratz, 2023-05-22:
I did build the mavsdk in debug mode.
Also the example.

ygratz, 2023-06-05:
[tlapik123](https://github.com/tlapik123) hi,
can you be more specific on how you use the fetchContent?

tlapik123, 2023-06-05:
> [tlapik123](https://github.com/tlapik123) hi, can you be more specific on how you use the fetchContent?

Sure! Here is the code.

```cmake
include(FetchContent)

set(PACKAGE_NAME mavsdk)

FetchContent_Declare(
    ${PACKAGE_NAME}
    GIT_REPOSITORY https://github.com/mavlink/MAVSDK.git
    GIT_TAG 3cd9aa992b8defe9687db3a6b71964363cd0f63b
)

FetchContent_GetProperties(${PACKAGE_NAME})
if(NOT ${PACKAGE_NAME}_POPULATED)
    FetchContent_Populate(${PACKAGE_NAME})

    add_subdirectory(${${PACKAGE_NAME}_SOURCE_DIR} ${${PACKAGE_NAME}_BINARY_DIR} EXCLUDE_FROM_ALL)
endif()
```
You can use the latest commit instead of commit provided.

You can include this in your *CMakeLists.txt* directly or you can create a separate *.cmake* file somewhere and include it with `include(<relative path to .cmake file)`

Then in the *CMakeLists.txt* you need to link against it.

```cmake
# link against mavsdk
target_link_libraries(<your target here>
    PRIVATE mavsdk
)
```

closed 2024-05-27 by julianoes