mavlink/MAVSDK #2045: "-DHTTP_ONLY=ON" in third_party/curl/CMakeLists.txt

tlapik123, 2023-05-22:
Is this intentional?

Since if the curl is build without https support the https .xml definition files sent by the camera will error out.
When the support it switched on it download the .xml correctly.

But maybe there is some issue for this that I have missed that explained why it is this way? :)

JonasVautherin, 2023-05-22:
I don't remember... probably it's just that OpenSSL is (was?) not a dependency if curl doesn't use HTTPS.

But then if you want to use HTTPS, you need a valid certificate on the drone. Do you have that?

tlapik123, 2023-05-22:
> I don't remember... probably it's just that OpenSSL is (was?) not a dependency if curl doesn't use HTTPS.
Is there any reason against using openSSL in this case? Less space/code?

> But then if you want to use HTTPS, you need a valid certificate on the drone. Do you have that?
I'm developing a test application and running that on the PC with the camera connected (through ardupilot). So this is not an issue for me.

JonasVautherin, 2023-05-22:
> Is there any reason against using openSSL in this case? Less space/code?

If it's not needed, it's always nice to have fewer dependencies. Also for our superbuild scheme, it's faster to build.

JonasVautherin, 2023-05-22:
Yeah it is used for camera_definition.xml files served as http :+1: