mavlink/MAVSDK #2047: MavlinkPassthrough low frequency callbacks

StefanoColli, 2023-05-22:
I'm using MavlinkPassthrough to subscribe to some topics, for example:
`mavlinkpassthrough.subscribe_message_async(MAVLINK_MSG_ID_POSITION_TARGET_LOCAL_NED, setpoint_callback)`
The callback is triggered only about 600 times on a 30 seconds flight (so at around 20Hz), while using MAVLink Inspector from QGC i can see that the topic is published at 50Hz. What am I missing?

julianoes, 2023-05-22:
Interesting. Are you checking this on the same link, so would it be the exact same number of messages?
I'm asking because different links from PX4 can send at different rates.

julianoes, 2023-05-23:
Right, so the rate might indeed be different. You can try to set the rate using the command https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL.

Or you can set it on the PX4 side here:
https://github.com/PX4/PX4-Autopilot/blob/main/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L26
using:

```
mavlink stream -r 50 -s POSITION_TARGET_LOCAL_NED -u $udp_offboard_port_local
```

StefanoColli, 2023-05-29:
By setting the rate with the command MAV_CMD_SET_MESSAGE_INTERVAL, I now receive the expected number of messages.
Thank you very much for your time.

closed 2023-05-29 by StefanoColli