mavlink/MAVSDK #2058: MissionRaw stopped working from v1.2.0 to v1.3.0

prokas-nikos, 2023-05-26:
In my android app after uploading from io.mavsdk:mavsdk-server:1.2.1 to io.mavsdk:mavsdk-server:1.3.1 I noticed that I could not upload the mission. After downgrading to 1.2.1 the functionality was the same as before.

I could upload with 
    implementation 'io.mavsdk:mavsdk:1.2.0'
    implementation 'io.mavsdk:mavsdk-server:1.2.1'

and could not upload with 
    implementation 'io.mavsdk:mavsdk:1.3.0'
    implementation 'io.mavsdk:mavsdk-server:1.3.1'

In the log I am receiving a warning "W  Invalid sequence for ArduPilot items"  and in Wireshark I can not see any mavlink messages sent. The MISSION_COUNT message is not sent.

JonasVautherin, 2023-05-26:
Are you using Ardupilot?

prokas-nikos, 2023-05-27:
I am using Ardupilot. 

I have found that in C++. I downloaded and found that piece of code. The part that I cannot understand is that with the method of implementation of mission with MAVSDK, it makes no sense for that piece of code to make any trouble since no one has to add the mission_count message in mavlink. It is added automatically automatically by mavsdk.

I will past the mission plan with all elements I am using.
`
private val MAV_FRAME_MISSION = 2
private val MAV_FRAME_GLOBAL_RELATIVE_ALT = 3
private val MAV_FRAME_GLOBAL_TERRAIN_ALT = 10

private val MAV_CMD_NAV_WAYPOINT = 16
private val MAV_CMD_NAV_RTL = 20
private val MAV_CMD_NAV_TAKEOFF = 22
private val MAV_CMD_CONDITION_YAW = 115
private val MAV_CMD_DO_CHANGE_SPEED = 178
private val MAV_CMD_DO_SPRAYER = 216
private val MAV_CMD_DO_SET_SERVO = 183

val missionItems : ArrayList<MissionRaw.MissionItem> = ArrayList()
var seq = 0

missionItems.add(
    MissionRaw.MissionItem(seq++,
        MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_TAKEOFF,
        1, 1,
        0.0F, 0.0F, 0.0F, Float.NaN,
        0, 0, alt,
        0)
)

missionItems.add(
    MissionRaw.MissionItem(seq++,
        MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_TAKEOFF,
        0, 1,
        0.0F, 0.0F, 0.0F, Float.NaN,
        (currentPos.latitude * 10.0F.pow(7)).toInt(), (currentPos.longitude * 10.0F.pow(7)).toInt(), alt,
        0)
)

for( i in waypoints.indices)
{
    //Set Steady Heading for each mission waypoint
    missionItems.add(
        MissionRaw.MissionItem(seq++,
            MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_CONDITION_YAW,
            0, 1,
            90.0F - activityViewModel.angleProgress.value!!.toFloat(), 0.0F, 0.0F, 0.0F,
            0, 0, 0.0F,
            0)
    )

    // Add Relative Waypoint to Mission
    missionItems.add(
        MissionRaw.MissionItem(seq++,
            MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_NAV_WAYPOINT,
            0, 1,
            0.0F, 0.0F, 0.0F, Float.NaN,
            (waypoints[i].latitude * 10.0F.pow(7)).toInt(), (waypoints[i].longitude * 10.0F.pow(7)).toInt(), alt,
            0)
    )

    // After First Mission waypoint added
    if(i == 0)
    {
        // Set Sprayer Enable ON
        missionItems.add(
            MissionRaw.MissionItem(seq++,
                MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_DO_SPRAYER,
                0, 1,
                1.0F, 0.0F, 0.0F, 0.0F,
                0, 0, 0.0F,
                0)
        )

        // Set Servo ON
        missionItems.add(
            MissionRaw.MissionItem(seq++,
                MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_DO_SET_SERVO,
                0, 1,
                5.0F, sprayerIntensityPWM, 0.0F, 0.0F,
                0, 0, 0.0F,
                0)
        )

        // Set user preferable speed in m/s
        missionItems.add(
            MissionRaw.MissionItem(seq++,
                MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_DO_CHANGE_SPEED,
                0, 1,
                1.0F, activityViewModel.flightSpeed.value!!.toFloat(), 0.0F, 0.0F,
                0, 0, 0.0F,
                0
            )
        )
    }
}

// Set Sprayer Enable OFF
missionItems.add(
    MissionRaw.MissionItem(seq++,
        MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_DO_SPRAYER,
        0, 1,
        0.0F, 0.0F, 0.0F, 0.0F,
        0, 0, 0.0F,
        0)
)

// Set Servo OFF
missionItems.add(
    MissionRaw.MissionItem(seq++,
        MAV_FRAME_GLOBAL_TERRAIN_ALT, MAV_CMD_DO_SET_SERVO,
        0, 1,
        5.0F, 1000.0F, 0.0F, 0.0F,
        0, 0, 0.0F,
        0)
)

// RTL - Return To Land
missionItems.add(
    MissionRaw.MissionItem(seq,
        MAV_FRAME_MISSION, MAV_CMD_NAV_RTL,
        0, 1,
        0.0F, 0.0F, 0.0F, 0.0F,
        0, 0, 0.0F,
        0)
)
`

julianoes, 2023-05-29:
We need to make a docs entry of how the sequence need to be for ArduPilot (and PX4).

The 0th item is home (or empty). The 1st item is the "first" waypoint and has sequence 0. Can you try that?

prokas-nikos, 2023-05-29:
The above implementation is the one that I have found out that it works with a lot of trial and error. 
If I understand correctly what you are saying is to replace the first element on my list
(this one)
 missionItems.add(
        MissionRaw.MissionItem(seq++,
            MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_TAKEOFF,
            1, 1,
            0.0F, 0.0F, 0.0F, Float.NaN,
            0, 0, alt,
            0)
    )

with this?
 missionItems.add(
        MissionRaw.MissionItem(seq++,
            MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_WAYPOINT,
            1, 1,
            0.0F, 0.0F, 0.0F, Float.NaN,
            curPos.latitude, curPos.longitude, alt,
            0)
    )

and after that the rest of the elements should be the same? So after the 0th element which will be MAV_CMD_NAV_WAYPOINT there should be a MAV_CMD_NAV_TAKEOFF as 1st element? This should function on both 1.2.0 and 1.3.0?

prokas-nikos, 2023-05-29:
I have changed the first element to the following
`        missionItems.add(
            MissionRaw.MissionItem(seq++,
                MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_WAYPOINT,
                1, 1,
                0.0F, 0.0F, 0.0F, Float.NaN,
                (currentPos.latitude * 10.0F.pow(7)).toInt(), (currentPos.longitude * 10.0F.pow(7)).toInt(), alt,
                0)
        )`

In v1.2.0 it still works and I can upload the mission but in v1.3.0 it fails to upload still with " W  Invalid sequence for ArduPilot items"

julianoes, 2023-05-29:
> with this for 0th element?
missionItems.add( MissionRaw.MissionItem(seq++, MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_CMD_NAV_WAYPOINT, 1, 1, 0.0F, 0.0F, 0.0F, Float.NaN, curPos.latitude, curPos.longitude, alt, 0) )

I think you don't want `seq++` here. The sequence should be: 0, 0, 1, 2, 3, ... Can you make sure that's what you're sending?

julianoes, 2023-05-29:
I might be wrong on this and I should make a simple Python example to show it. It's clearly a bit of challenge.

prokas-nikos, 2023-05-30:
ok. I found the issue. My code was working on version 1.2.0 but stopped working on v1.3.0. After @julianoes last comment I started digging into Python examples and found the mission_import.py example. I added a loop to print out the elements of the example-mission.plan and noticed the correct sequence. My problem was that even when I tried with the correct sequence(0, 0, 1, 2, 3 ...) I was not using the correct sequence with current and autocontinue as shown below in example-mission.plan. The current needs to be 0 in the first mission_item(Waypoint:16) and then 1 in the second waypoint(Takeoff:22)

MissionItem: [seq: 0, frame: 5, command: 16, current: 0, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: 0.0, x: 473977420, y: 85455939, z: 488.7872009277344, mission_type: 0]
MissionItem: [seq: 0, frame: 3, command: 22, current: 1, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: nan, x: 473977420, y: 85455939, z: 50.0, mission_type: 0]
MissionItem: [seq: 1, frame: 3, command: 16, current: 0, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: nan, x: 473982266, y: 85455604, z: 50.0, mission_type: 0]
MissionItem: [seq: 2, frame: 3, command: 16, current: 0, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: nan, x: 473982493, y: 85449242, z: 50.0, mission_type: 0]
MissionItem: [seq: 3, frame: 3, command: 16, current: 0, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: nan, x: 473977416, y: 85449242, z: 50.0, mission_type: 0]
MissionItem: [seq: 4, frame: 2, command: 20, current: 0, autocontinue: 1, param1: 0.0, param2: 0.0, param3: 0.0, param4: 0.0, x: 0, y: 0, z: 0.0, mission_type: 0]  

When I applied the same paradigm on my code it works on both v1.2.0 and v1.3.0.

prokas-nikos, 2023-05-30:
Thank you for your support. I was wrong, it was not an issue after all.

ykhedar, 2023-06-02:
Hi @prokas-nikos,
The current sequence for mission sequence is 0,1,2,3, etc. We had incorrectly implemented ardupilot mission validation which has been now corrected and pushed to the main branch. I am not sure about the status in v1.3.0, but i believe that release was from much before the addition of ardupilot mission validation check. 

If you are using ArduPilot, I would suggest to use the main branch instead of specific tagged versions since many ardupilot related changes are in the main branch. After the release of v2.0 these should be available also in the release.

More information on how to setup and work with ArduPilot and MAVSDK are in this repository https://github.com/ykhedar/ardupilot-mavsdk which I specifically created to help devs like you get started with this combination. Give it a try and let me know if you face any problem.

julianoes, 2023-06-04:
> I am not sure about the status in v1.3.0, but i believe that release was from much before the addition of ardupilot mission validation check.

I backported it: https://github.com/mavlink/MAVSDK/pull/2065

It's part of [v1.4.15](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.15)

I need to update the Python wrappers to be based on top of it.

julianoes, 2023-06-04:
@JonasVautherin can you update the Java wrappers as well? Thanks.

prokas-nikos, 2023-06-07:
Hi @ykhedar, 
thanks for the info. I will check out for sure. The reason I moved to previous versions was to locate where the issue came. I am using MAVSDK one year now, and when I tried to make some changes to my app and upgrade to the latest version of MAVSDK the problem appeared.

julianoes, 2023-06-07:
This has been fixed in MAVSDK [v1.4.15](https://github.com/mavlink/MAVSDK/releases/tag/v1.4.15), MAVSDK-Python [v1.4.7](https://github.com/mavlink/MAVSDK-Python/releases/tag/1.4.7), and MAVSDK-Java [1.3.2](https://github.com/mavlink/MAVSDK-Java/releases/tag/1.3.2).

Thanks @ykhedar and @JonasVautherin.

closed 2023-06-07 by julianoes