mavlink/MAVSDK #2059: Does the MAVSDK support mavlink ftp server

tbago, 2023-05-27:
Hi support,
  I know there are [ftp server example](https://github.com/mavlink/MAVSDK/tree/main/examples/ftp_server). I try to use this example but seems not working.
  
  I read some **mavskd proto code**. And found the plugins have pair server and client side( e.g.: **camera -> camera_server**, **param->param_server**). But for ftp there is only ftp proto (seems to work as client).  So if I want to support ftp server, I need to add the ftp_server.proto. And implment it as a server component.

  Is my understanding correct or  just am I missing something in **MAVSDK**?

JonasVautherin, 2023-05-28:
Right now, both clients and servers are using `ftp.proto`. But there is ongoing work to split that into something like `ftp.proto` and `ftp_server.proto`.

Still the examples should work. What do you mean "they seem to not be working"? Did you compile and try them?

julianoes, 2023-05-29:
It's work in progress: https://github.com/mavlink/MAVSDK/pull/2060

If you need it urgently, consider my consulting services.

julianoes, 2024-05-27:
Has been done for a while now.

closed 2024-05-27 by julianoes