mavlink/MAVSDK #2063: can't connect to SITL - udp

ygratz, 2023-05-30:
I try connecting to a SITL using takeoff_and_land example. I use udp to 127.0.0.1:14540

1. For some reason I get the following fault:
MAVSDK version: v0.37.0 (mavsdk_impl.cpp:21)
Trying Connection: udp://:14540
Waiting to discover system...
[04:40:14|Info ] New system on: 127.0.0.1:45563 (with sysid: 1) (udp_connection.cpp:190)
[04:40:14|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:418)
[04:40:14|Debug] Component Autopilot (1) added. (system_impl.cpp:324)
[04:40:15|Warn ] sending again after 0.507609 s, retries to do: 3  (520). (mavlink_commands.cpp:239)
[04:40:15|Warn ] sending again after 1.01592 s, retries to do: 2  (520). (mavlink_commands.cpp:239)
[04:40:16|Warn ] sending again after 1.51982 s, retries to do: 1  (520). (mavlink_commands.cpp:239)
[04:40:16|Error] Retrying failed (520) (mavlink_commands.cpp:260)
No autopilot found.

2. Where is mavlink_commands.cpp located?

ygratz, 2023-05-30:
On the SITL I get the following, when trying to connect:

Connection reset or closed by peer on TCP socket
Attempting reconnect
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
Attempting reconnect
...

julianoes, 2024-05-27:
No follow up, closing.

closed 2024-05-27 by julianoes