mavlink/MAVSDK #2068: Installing and building MAVSDK for debugging

ygratz, 2023-06-01:
Hi,
After having various issues with debugging, I decided to install and build MAVSDK from scratch.
I followed the instructions on:
https://mavsdk.mavlink.io/main/en/cpp/guide/installation.html
and installed version: 1.4.13

Then followed the instructions on:
https://mavsdk.mavlink.io/main/en/cpp/guide/build.html
Local Install
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -H.
cmake --build build/default --target install

Then I tried building the examples:
cd examples
cmake -DCMAKE_PREFIX_PATH=$(pwd)/../install -Bbuild -H.
cmake --build build -j8

I get the following message:
**CMake Error at /usr/share/cmake-3.22/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "FindMAVLink.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "MAVLink", but
  CMake did not find one.

  Could not find a package configuration file provided by "MAVLink" with any
  of the following names:

    MAVLinkConfig.cmake
    mavlink-config.cmake

  Add the installation prefix of "MAVLink" to CMAKE_PREFIX_PATH or set
  "MAVLink_DIR" to a directory containing one of the above files.  If
  "MAVLink" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /home/ygratz/MAVSDK/install/lib/cmake/MAVSDK/MAVSDKConfig.cmake:3 (find_dependency)
  autopilot_server/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!**

Please advise.

*If there are different instructions which I need to follow to install MAVSDK for debugging - please refer me to it.
Thank you.

Note: I do see that "include" dir was created and it does include various files. See attached picture.
[image]

ygratz, 2023-06-01:
Now even more bizzare is that I'm able to build and run takeoff_and_land example.
The only difference I can think about was that following:
https://mavsdk.mavlink.io/main/en/cpp/quickstart.html

After getting the MAVSDK version from the git, I did:
To build the takeoff and land example, you can do:

cd takeoff_and_land/
cmake -Bbuild -H.
cmake --build build -j4

ygratz, 2023-06-01:
This was before performing the Local Install (as described)

JonasVautherin, 2023-06-01:
You built MAVSDK v1.4.13, but you are trying to build the examples from `main`. You should build the examples from the `v1.4.13` tag :+1:

tbago, 2023-06-01:
You can try this command first to install the mavlink header to system.
```
cmake -Bthird_party/mavlink/build -Sthird_party/mavlink
sudo cmake --build third_party/mavlink/build
```
For this [commit](https://github.com/mavlink/MAVSDK/commit/04e78f163d3dd630103a353cca8d04fd943f8486) they have separate the mavlink and mavsdk.

JonasVautherin, 2023-06-01:
> For this [commit](https://github.com/mavlink/MAVSDK/commit/04e78f163d3dd630103a353cca8d04fd943f8486) they have separate the mavlink and mavsdk.

That's for the `main` branch, but v1.4.13 doesn't have that change yet.

ygratz, 2023-06-01:
[JonasVautherin](https://github.com/JonasVautherin) Thanks!
I understood that I got the code from main() instead of v1.4.13.
So I did the following:
1. wget https://github.com/mavlink/MAVSDK/releases/download/v1.4.13/libmavsdk-dev_1.4.13_ubuntu20.04_amd64.deb
and unpacked it.
2. I got the MAVSDK.zip from:
https://github.com/mavlink/MAVSDK/tree/v1.4.13

3. I followed the build instructions:
cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.

However, now I get the following:

CMake Error at src/mavsdk/core/CMakeLists.txt:51 (set_target_properties):  set_target_properties called with incorrect number of arguments.

CMake Error at src/CMakeLists.txt:63 (add_subdirectory):
  The source directory
    /home/ygratz/MAVSDK/src/third_party/gtest
  does not contain a CMakeLists.txt file.

Now - what am I missing here.
I just try to take a released version and compile it for debug purpose.
Shall I take a different tag?

ygratz, 2023-06-04:
Hi,
I removed the MAVSDK package (libmavsdk-dev_1.4.13_ubuntu20.04_amd64.deb) and erased the MAVSDK folder that I had.
I also checked /usr/ for any "mavsdk" files. 

Then I re-installed the package and downloaded  the MAVSDK folder of version 1.4.13.
I still get the same issue.

Is there something specific to look for when re-installing the MAVSDK? that I need to verify that I'm erasing.

tbago, 2023-06-04:
```
/home/ygratz/MAVSDK/src/third_party/gtest
does not contain a CMakeLists.txt file.
```
Seems like you don't recursive download submodule, please check the folder **/home/ygratz/MAVSDK/src/third_party/gtest** first.

ygratz, 2023-06-04:
Hi,
when I do:
git clone https://github.com/mavlink/MAVSDK.git
it clones the main branch of MAVSDK,

Now, can I instead clone a given release version. Lets say 1.4.15?

ygratz, 2023-06-04:
How

tbago, 2023-06-04:
You can just use the following command to clone **recursive**. and swith to tag v1.4.15
```
git clone https://github.com/mavlink/mavsdk --recursive
cd mavsdk
git checkout v1.4.15
```

ygratz, 2023-06-05:
[tbago](https://github.com/tbago) hi,
are you using VSCode?
and if so, when you debug - are you able to step-into mavsdk functions?

Following the build instructions - I am able to build, run and debug an example.
But it doesn't step-into a library function.

tbago, 2023-06-05:
I use the gdb to debug the source code. It works well. 

So first you can use gdb to check the **step into** works well.

ygratz, 2023-06-13:
Thanks.
I did eventually manage to solve the problem.
I had multiple instances of libmavsdk.so. One was in release mode, and it was the one referenced by my debugger.

closed 2023-06-13 by julianoes