mavlink/MAVSDK #2070: mavsdk server compile failed on MacOS 11.6

tbago, 2023-06-01:
Hi support:
I use the following command to compile the mavsdk with server support:
```
cmake -B ./build -DCMAKE_EXPORT_COMPILE_COMMANDS=1 -S . -DBUILD_MAVSDK_SERVER=1 
```
I got the following link error:
Undefined symbols for architecture x86_64:
```
  "_SSL_get_peer_certificate", referenced from:
      ssl_handshaker_result_extract_peer(tsi_handshaker_result const*, tsi_peer*) in libgrpc.a(ssl_transport_security.cc.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
This symbols error is about the **openssl** version. I already compile the offical grpc, they use the **openssl1.1**.
The mavsdk use the openssl version is **[3.0.5](https://github.com/mavlink/MAVSDK/blob/main/third_party/openssl/CMakeLists.txt#LL8C5-L8C20)**. 
In the new openssl include file the **SSL_get_peer_certificate** function is deprecated.
```
/* Deprecated in 3.0.0 */
#  ifndef OPENSSL_NO_DEPRECATED_3_0
#   define SSL_get_peer_certificate SSL_get1_peer_certificate
#  endif
# endif
```
Does the SSL_get_peer_certificate still working in openssl version 3.0.5.

julianoes, 2023-06-01:
Hm ok, so what do you suggest we should do?

tbago, 2023-06-02:
Hi , I just change the openssl version 1.1 to build the project. I add some platform special code in the **[CMakeList.txt](https://github.com/mavlink/MAVSDK/blob/main/third_party/openssl/CMakeLists.txt#L43)**

```
elseif(CMAKE_HOST_APPLE)
    set(OPENSSL_VERSION 1.1.1o)
    ExternalProject_add(
        openssl
        URL https://www.openssl.org/source/openssl-${OPENSSL_VERSION}.tar.gz
        PREFIX openssl
        CONFIGURE_COMMAND <SOURCE_DIR>/config ${OPENSSL_BUILD_TYPE} --prefix=${CMAKE_INSTALL_PREFIX} no-shared
        )
```

This may be related to the computer operating system version.  I'm not sure if anyone else has the same problem.

closed 2023-06-02 by tbago

JonasVautherin, 2023-06-02:
> The mavsdk use the openssl version is [3.0.5](https://github.com/mavlink/MAVSDK/blob/main/third_party/openssl/CMakeLists.txt#LL8C5-L8C20).

Both MAVSDK and gRPC should use the version found by CMake, which is OpenSSL v3 when using the superbuild.

Now sometimes there are issues with the superbuild on some platforms, so you would have to check the full build log (specifically around the build of openssl).