mavlink/MAVSDK #2075: SIGABRT when destructing mavsdk

jperez-droneup, 2023-06-09:
In order to start fresh after detecting a disconnect (we want to reconnect on a potentially different IP address), we have to destruct the mavsdk object and create a brand new one. However, I have seen 2 exceptions come up when destructing the mavsdk.

I have attached a CMakeLists.txt and relevant minimal examples. Change the source `.txt` files to `.cpp`. Here is the CMake line:
`cmake -DCMAKE_INSTALL_PREFIX=install -DCMAKE_BUILD_TYPE=Debug ..`

It is also worth noting that I am connecting to the simulator using a remote port and UDP. Usage is similar to `mavsdk_server`:
`case1 udp://<url>:14540`. I am also using a REQUIRED code change from [this PR](https://github.com/mavlink/MAVSDK/pull/2074) to allow the UDP disconnect to be properly recognized.

1. When registering on the `subscribe_is_connected` and then killing the simulator after a successful connection, I get a SIGABRT on [this line](https://github.com/mavlink/MAVSDK/blob/ea0b720538cff8e51947557234cb9c4300edadc3/src/mavsdk/core/mavsdk_impl.cpp#L53). I made sure to unsubscribe from all subscriptions before destructing the mavsdk, but it still seems to be holding onto something. The error on screen is `libc++abi: terminating due to uncaught exception of type std::__1::system_error: thread::join failed: Resource deadlock avoided`. I have attached this as case1.cpp. This code gives you 120 seconds to kill the simulator and detect the failure.
2. After the mavsdk is destructed, it looks like the system_thread is still running, but the `_parent.time` has been uninitialized and leads to a `EXC_BAD_ACCESS (code=1, address=0x10)` exception on [this line](https://github.com/mavlink/MAVSDK/blob/ea0b720538cff8e51947557234cb9c4300edadc3/src/mavsdk/core/mavsdk_time.cpp#L45).
[CMakeLists.txt](https://github.com/mavlink/MAVSDK/files/11702438/CMakeLists.txt)
[case2.txt](https://github.com/mavlink/MAVSDK/files/11702441/case2.txt)
[case1.txt](https://github.com/mavlink/MAVSDK/files/11702442/case1.txt)

JonasVautherin, 2023-06-09:
Which version of MAVSDK are you using?

jperez-droneup, 2023-06-09:
I am using my PR branch, which is based off of today's trunk `MAVSDK version: v1.4.0-484-gc1006bf3`. I was using a slightly older version previously, it looks like case2 has been fixed since then. However case1 is still present.

julianoes, 2023-06-09:
1. Oh, I see the problem. You are trying to destruct the `Mavsdk` instance while you are inside the callback. That can't really work. I took the inspiration from your case1 and created an example: https://github.com/mavlink/MAVSDK/pull/2078 Let me know if that could work for you.

julianoes, 2023-06-09:
2. This one runs fine for me:

```
[11:29:51|Info ] MAVSDK version: v1.4.0-484-gf8f3daa6 (mavsdk_impl.cpp:23)
[11:29:51|Info ] New system on: 127.0.0.1:14580 (with system ID: 1) (udp_connection.cpp:192)
[11:29:51|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:671)
[11:29:51|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[11:29:51|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:222)
[11:29:51|Debug] Discovered 1 component(s) (system_impl.cpp:548)
Connection successful
[11:29:51|Debug] MAVLink: info: [logger] ./log/2023-06-09/23_29_49.ulg	 (system_impl.cpp:248)
Destructing mavsdk
Destruction successful
Constructing new mavsdk
[11:30:06|Info ] MAVSDK version: v1.4.0-484-gf8f3daa6 (mavsdk_impl.cpp:23)
[11:30:06|Info ] New system on: 127.0.0.1:14580 (with system ID: 1) (udp_connection.cpp:192)
[11:30:06|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:671)
[11:30:06|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[11:30:07|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:222)
[11:30:07|Debug] Discovered 1 component(s) (system_impl.cpp:548)
Reconnection successful.
```

JonasVautherin, 2023-06-09:
Yeah, seems like @jperez-droneup observed that [above](https://github.com/mavlink/MAVSDK/issues/2075#issuecomment-1584820336):

> it looks like case2 has been fixed since then. However case1 is still present.

julianoes, 2023-06-10:
Alternatively to completely having to destruct and re-instantiate Mavsdk, you could probably just remove the connection and add it again, right?

I'm adding support for that here: https://github.com/mavlink/MAVSDK/pull/2079

jperez-droneup, 2023-06-10:
Indeed this was a user error - I didn't realize that the thread being joined was the thread causing the destruction. But yes, having a way to remove the connection without having to destruct the mavsdk at all is definitely better. Thanks for your fast responses!

julianoes, 2023-06-10:
Fixed with #2077, #2078, and #2079.

closed 2023-06-10 by julianoes