mavlink/MAVSDK #2076: UDP to remote port does not recognize disconnect

jperez-droneup, 2023-06-09:
Configuring the mavsdk to `always_send_heartbeats` will properly send heartbeats to the simulator/vehicle, however [this line](https://github.com/mavlink/MAVSDK/blob/ea0b720538cff8e51947557234cb9c4300edadc3/src/mavsdk/core/mavsdk_impl.cpp#L536) sets the system to always consider itself as connected. So even though the mavsdk is sending heartbeats, if enough are missed to normally signal a disconnect, that disconnect is never recognized and any `subscribe_is_connected` callbacks on the system are not fired. I raised https://github.com/mavlink/MAVSDK/pull/2074 to fix this issue.

JonasVautherin, 2023-06-09:
Oh, that seems like a bug indeed.

BTW, no need to split your PR into an issue and a PR, that just splits the description (in this care I reviewed the PR without the context of this issue :see_no_evil:).

If you want to raise an issue (but have no PR for it), then open an issue. If you want to open a PR directly, feel free to do it, no need to create a side-issue just for the sake of it :wink:.

jperez-droneup, 2023-06-09:
Although I am a seasoned enough engineer, I am fairly new to github >< Thanks for letting me know. Whichever is more helpful for you. I have answered your questions on the PR.

PR #2077 (2023-06-10): core: remove always connected flag
This flag was added when we didn't have the notion of a component, and therefore it was required for the FTP server implementation. I believe we should be able to remove it now.

Fixes #2076.
closed 2023-06-10 by julianoes