mavlink/MAVSDK #2082: Compatibility to Ardupilot version 4.0.5

ygratz, 2023-06-13:
Hi,
I have a project in which I need to connect and fly a quadcopter running Ardupilot version 4.0.5.
Is it possible to use MAVSDK for that?

If not, shall I use the MAVLink libraries instead?
https://mavlink.io/en/getting_started/use_libraries.html
(although it is stated they are used on MAVSDK)

JonasVautherin, 2023-06-13:
There is some support for Ardupilot in MAVSDK, so you can try and see if what you need is supported :blush:.

Also we happily take contributions for Ardupilot support!

holden-zenith, 2023-06-13:
@JonasVautherin If you guys actively keep up with PRs then I'd be willing to make significant progress with ardupilot support

JonasVautherin, 2023-06-13:
We do! The `MissionRaw` plugin works (I know it's being used), and I think it would not require too much to get the `Mission` plugin to work with Ardupilot. Would be great if you could look into that!

ykhedar, 2023-06-21:
Hi @ygratz, just try the examples at https://github.com/ykhedar/ardupilot-mavsdk. In the docker file for Ardupilot SITL, you can  build the 4.0.5 version of ArduCopter and run the mavsdk mission examples to check if it works. Most probably it should.

Hi, @holden-zenith we use Ardupilot with MAVSDK actively in our company so you can count on continued support from my side as well. Have you some specific feature support in mind?

JonasVautherin, 2023-06-21:
Feel free to share even snippets of your mavlink-passthrough code. Maybe somebody will contribute a PR eventually :pray:.

Jai-GAY, 2023-08-10:
For physical Ardupilot communication, I have successfully tested and worked well with the[ Andruav_ap Android app](https://play.google.com/store/apps/details?id=arudpilot.andruav&hl=en_SG&gl=US). [Github](https://github.com/HefnySco/andruav_android_app)

I hope it helps someone struggling.

julianoes, 2024-05-27:
I'm not so sure what this is about any more. I'll close it. If I'm wrong, just re-open and convince me otherwise.

closed 2024-05-27 by julianoes