mavlink/MAVSDK #2083: Mavlink header install using ExternalProject_Add

jperez-droneup, 2023-06-13:
When building MAVSDK using CMake's ExternalProject_Add, with SUPERBUILD on, it will properly pull and build all of MAVSDK's dependencies. However, I get errors for missing mavlink headers and indeed [this commit](https://github.com/mavlink/MAVSDK/commit/04e78f163d3dd630103a353cca8d04fd943f8486#diff-4c224720c825bc0b47c1375867b1f0eb0f7943639c4ef77d1d7ef38e9792ccb6) changed where the files were installed to. It would be great if we could either have the mavlink headers installed alongside the mavsdk ones automatically, or at least have a CMake option to change where dependencies are installed to.

I would suggest possibly setting `DEPS_INSTALL_PATH` to `CMAKE_INSTALL_PREFIX` on [this line]( https://github.com/mavlink/MAVSDK/blob/0338c3e5eb86ec6c00a11ae99b019c97646b907a/CMakeLists.txt#L56) . Alternatively, you could make `DEPS_INSTALL_PATH` an optional argument, so users can place it to their needs, and it would be accessible via ExternalProject_Add.

JonasVautherin, 2023-06-13:
This said, after some experience with CMake superbuilds, my more recent opinion is that we should provide helpers, but not a superbuild (see my blog post [here](https://www.acarg.ch/posts/cmake-deps/)). That's much less confusing in the sense that it makes it clear to the user that they should handle their dependencies.

But well, doesn't change the fact that MAVSDK uses this superbuild, and many users don't want to know about how they should handle their dependencies :see_no_evil:. So probably your best bet is what I wrote in my previous message.

jperez-droneup, 2023-06-13:
Yeah, for the time being I am just using:
```
target_include_directories(my-project PUBLIC
    ${CMAKE_BINARY_DIR}/mavsdk/src/mavsdk_proj-build/third_party/install/include
)
```
An option would be nice to have still, just so it's not quite so hacky.

JonasVautherin, 2023-06-13:
Just build your project with `CMAKE_PREFIX_PATH=path/to/mavsdk/src/mavsdk_proj-build/third_party/install`, right? That would be clean.

jperez-droneup, 2023-06-13:
To a degree, yeah, if I wanted to put mavlink specifically into my cmake code. I was trying to keep it generic. When using ExternalProject_Add, the code is not yet in CMake at generation time, so I can't use find_package on a target which is not installed yet. If I pulled the full repo and had it as a submodule instead (before calling any CMake), then yes I could use the CMAKE_PREFIX_PATH.

julianoes, 2023-06-14:
I somewhat reverted this to the previous behaviour: #2084.

JonasVautherin, 2023-06-14:
> To a degree, yeah, if I wanted to put mavlink specifically into my cmake code. I was trying to keep it generic.

Well MAVSDK transitively depends on MAVLink in its public interface (before #2084), and therefore `find_package` must find MAVLink somewhere. It's not about "putting mavlink specifically into your cmake code", it won't actually change your CMake code. You just have to tell your CMake project where to find the dependencies with `find_package`. Either you install MAVLink on your system, or you point CMAKE_PREFIX_PATH to it.

When you install the MAVSDK `.deb` package, it depends on stuff like openssl, and it will require you to install them on your system. Still you don't consider that you are "putting openssl specifically into your cmake code", right? That's exactly the same for MAVLink, except that MAVLink is not available with `apt install libmavlink-dev`.

> I somewhat reverted this to the previous behaviour: https://github.com/mavlink/MAVSDK/pull/2084.

Yeah I guess that solves the confusion around the transitive dependency :+1:

jperez-droneup, 2023-06-14:
I had tried adding to the CMAKE_PREFIX_PATH like you suggested @JonasVautherin, but still got the same header include errors. That's why I resorted to the more manual target_include_directories entry.

julianoes, 2023-06-14:
Try my PR/branch @jperez-droneup .

jperez-droneup, 2023-06-14:
Meant to comment, I tried it and works well, with no janky include paths :)

closed 2024-05-27 by julianoes