mavlink/MAVSDK #2086: Manual control while rc control connected

JacobAi, 2023-06-14:
I'm trying to use manual control for controlling drone. I manage to operate the drone in that mode as long as the rc remote is disconnected from the pixhawk.
The only way manual control is working is when I set the paramter COM_RC_IN_MODE to 3 (RC or Joystick keep first). 
Is there a way that manual control will work until there is a taking control from the rc controller (keep the rc controller connected)?

julianoes, 2023-06-14:
These are the options: https://docs.px4.io/v1.13/en/advanced_config/parameter_reference.html#COM_RC_IN_MODE

What is the manual control by MAVSDK doing in your case? Can you explain that? Potentially, you could consider using offboard for that instead to make it more of a distinction.

JacobAi, 2023-06-15:
The px4 ignores commands from mavsdk as long the rc connected to the pixhawk.

JacobAi, 2023-06-16:
I tried using offboard mode with velocity control. It's not precice and smooth like using manual control.
Can I ovveride rc controller commands with manual control till the rc controller change it's state?

julianoes, 2023-06-16:
If done right, it should be just as "precise". Again, it's not recommended to (mis)use manual control for autonomous control that can be overriden/switched out of by offboard control.

closed 2026-04-02 by julianoes