mavlink/MAVSDK #2089: Cmake error at third party build target

SImchaMalina, 2023-06-19:
I downloaded version 1.4 and unzipped it, however when I try to bulid it, I am getting this error.

[cmake] ModuleNotFoundError: No module named 'future'
[cmake] mingw32-make.exe[2]: *** [CMakeFiles\mavlink.dir\build.make:126: mavlink/src/mavlink-stamp/mavlink-configure] Error 1
[cmake] mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:95: CMakeFiles/mavlink.dir/all] Error 2
[cmake] mingw32-make.exe: *** [Makefile:103: all] Error 2
[cmake] CMake Error at third_party/cmake/build_target.cmake:51 (message):
[cmake]   C:/MAVSDK-1.4/build/Release/third_party/mavlink failed to
[cmake]   build!
[cmake] Call Stack (most recent call first):
[cmake]   third_party/CMakeLists.txt:31 (build_target)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!

What exactly is going wrong here?

JonasVautherin, 2023-06-19:
> ModuleNotFoundError: No module named 'future'

It's not finding the `future` module of python. Did you try installing it?

SinPingWu, 2024-02-05:
I also have the same problem and the future module is installed, how can I solve this problem.
[image]

JonasVautherin, 2024-02-05:
I have seen situations where there are multiple versions of Python installed, and it detects one where the module is not installed.

Can you tried to run the build from a Python venv? Something like this:

```
python3 -m venv venv # create the venv
source ./venv/bin/activate # get into the venv
pip install future # install future in the venv
cmake [...] # run your MAVSDK build in the venv, with the cmake command you usually use
```

SinPingWu, 2024-02-06:
I solved the problem by rebuilding the dockcross/android-arm64 image.

`
FROM dockcross/android-arm64:latest

ENV DEFAULT_DOCKCROSS_IMAGE mydockcross/android-arm64
RUN apt-get install python3-future
`

JonasVautherin, 2024-02-06:
Oh you are using dockcross. The original issue was on Windows.

I am a bit surprise that it doesn't work for you on dockcross, because it works [in our CI](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L337-L379) :thinking: