mavlink/MAVSDK #2095: Different connections to the same system

eastman-bechamo, 2023-06-28:
I have a MAVLink-enabled autopilot that I'm connecting to from a PC using MAVSDK. I'd like to communicate with the AP using two different connections, one USB and one UART. I can connect to the AP using either one, but it doesn't seem that I'm able to connect to both at the same time.

Specifically, my approach is to call `add_any_connection` twice and get two different `System` pointers. From those pointers, I want to create a `MavlinkPassthrough` so that I can send custom MAVLink messages directly to the device.

Is this possible? If it is, what kind of approach should I take?

Thank you!

julianoes, 2023-06-28:
Hmm, that's not how the MAVSDK architecture works. A system is a representation for a MAVLink system. You only have one system, and two connections to it.

Why do you want to have two MavlinkPassthrough instances? Do you want outgoing messages to be sent on both connections (which it ought to do unless something is broken), or do you want to send specific messages on either one or the other connection?

eastman-bechamo, 2023-06-28:
Hi @julianoes , thanks for replying. If it's possible, I'd like to send some messages over one connection, and some over a different connection exclusively.

julianoes, 2023-06-29:
That's not really possible then. The only way to make that work would be to add an intercept_mavlink_messages() function per connection rather than per mavsdk instance. The problem with that would be that there is currently no public facing connection object, so it's not obvious how we would do it.

Maybe to take take a step back, what sort of filtering would you want to do? What's the idea behind it? Maybe we can add connection configuration based on something higher level, like the notion of a low bandwidth fallback link (e.g. satellite)?

JonasVautherin, 2023-06-29:
Are you trying to achieve some kind of multihoming? Like having redundancy between two different links (e.g. both a radio link and an Internet link), such that messages go through the "best" path?

eastman-bechamo, 2023-06-29:
@JonasVautherin @julianoes I appreciate the questions, but unfortunately that's as much detail as I can give right now :(

I'll admit that I'm not extremely familiar with the architecture of MAVSDK, but how hard would it be for a system to maintain and choose between two connections?

What does a `System` require of a connection after initialization? Is there any reason that initialization couldn't happen twice for the same `System` object?

Thank you :-)

eastman-bechamo, 2023-06-29:
@JonasVautherin I know the lack of information is hard, my apologies. What do you mean by "in front of MAVSDK"?

JonasVautherin, 2023-06-29:
Some kind of proxy like mavlink-router, running between the drone and the mavsdk instance. If that makes sense?

julianoes, 2023-06-30:
If you can't share why you would need a feature, that's unfortunately not quite enough incentive for me to implement/change anything to solve your challenge in your top secret project. :smirk:

julianoes, 2026-09-15:
Not sure if there is something actionable in here, closing.

closed 2026-09-15 by julianoes