mavlink/MAVSDK #2098: Add make uninstall

Zipdox2, 2023-07-11:
Currently you can make install but there's no make uninstall.

JonasVautherin, 2023-07-11:
If you install locally, then it's easy to delete. If you install globally on your system, I would suggest you rely on a package manager instead of doing something like `sudo make install` :+1:.

julianoes, 2023-07-11:
We are just using cmake. If we wanted to implement this, here is how: https://gitlab.kitware.com/cmake/community/-/wikis/FAQ#can-i-do-make-uninstall-with-cmake

I personally wouldn't bother but if you want to contribute it and maintain it going forward, feel free to make a PR.

JonasVautherin, 2023-07-11:
I would honestly vote against. I am always nervous about scripts that do any kind of `rm -rf $some_variable`, and I wouldn't want to be held responsible for removing random files on a user's system.

Also installing random stuff on your system with `sudo make install` is a bad idea. That is literally the job of the package manager, it should not be done with cryptic scripts in CMake (can you honestly assure me that `"@CMAKE_COMMAND@" ARGS "-E remove \"$ENV{DESTDIR}${file}\""` will never, ever remove stuff you did not expect?).

julianoes, 2023-07-11:
Right agreed. Let's stick with the default cmake setup that we have now.

closed 2023-07-11 by julianoes