mavlink/MAVSDK #2100: get params segmentation fault

bwjin310, 2023-07-19:
Hi, here are some questions when I using MAVSDK C++. I'm new to this, appreciated if helping me out 
- Use mavsdk::Telemetry plugin. When automatically retrieving certain parameters, such as "CAL_ACC0_ID", "CAL_MAG0_ID", a segmentation fault shows up. 
- Use mavsdk::Telemetry plugin. After receiving parammeters , trying to excute `mavsdk::TelemetryImpl::set_health_gyrometer_calibration(bool) ()` results in a instance of `std::system_error`
- Use mavsdk::Param plugin. When excuting `get_all_params()`, a `std::future_error `shows up.

## Environment

- MAVSDK version v1.4.13
- PX4 version v1.12.3

 ## Stacktrace

I subscribed the health messages simultaneously. When gps is active, the health message tells the local positon, global position and home position is all ok,. It seems under the circumstance, the sefmentation fault after retrieving "CAL_ACC0_ID", "CAL_MAG0_ID" can be observed, but the `std::system_error` not. I don't know if it's related, if not, just ignore this. 

1. GPS active

```
Got Health:	 

Gyro calibration:	 ok 

Accel calibration:	 ok 

Mag calibration:	 ok 

Local position:		 ok 

Global position:	 ok 

Home position:		 ok 

[04:39:00|Debug] getting param CAL_MAG0_ID, extended: no (mavlink_parameters.cpp:371)

Thread 54 "MavsdkInit Thre" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffebffff700 (LWP 3098)]
0x000055555b115340 in ?? ()
(gdb) 
#0  0x000055555b115340 in ?? ()
#1  0x00007fffe1e76ce9 in mavsdk::MAVLinkParameters::get_param_int_async(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (mavsdk::MAVLinkParameters::Result, int)> const&, void const*, std::optional<unsigned char>, bool) ()
   from /home/jss/VT4Release/moduleLib/libmavsdk.so.1
#2  0x00007fffe1e215a5 in mavsdk::SystemImpl::get_param_int_async(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (mavsdk::MAVLinkParameters::Result, int)> const&, void const*, std::optional<unsigned char>, bool) ()
   from /home/jss/VT4Release/moduleLib/libmavsdk.so.1
#3  0x00007fffe1fd0de8 in mavsdk::TelemetryImpl::check_calibration() ()
   from /home/jss/VT4Release/moduleLib/libmavsdk.so.1
#4  0x00007fffe1e1a747 in mavsdk::CallEveryHandler::run_once() ()
   from /home/jss/VT4Release/moduleLib/libmavsdk.so.1
#5  0x00007fffe1e30311 in mavsdk::MavsdkImpl::work_thread() ()
   from /home/jss/VT4Release/moduleLib/libmavsdk.so.1
#6  0x00007fffee5ab6df in ?? ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fffe919e6db in start_thread (
    arg=0x7ffebffff700) at pthread_create.c:463
#8  0x00007fffee00661f in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

```
Got Health:	 

Gyro calibration:	 ok 

Accel calibration:	 ok 

Mag calibration:	 ok 

Local position:		 ok 

Global position:	 ok 

Home position:		 ok 
[... 183 lines cut ...]
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local calibration:	not ok
Global calibration:	not ok
Home calibration:	not ok
Got Health:	
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local calibration:	not ok
Global calibration:	not ok
Home calibration:	not ok
Got Health:	
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local calibration:	not ok
Global calibration:	not ok
Home calibration:	not ok
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: No associated state
```

When I use `get_all_params()`, I noticed that the Gyro calibration and Accel calibration turned into not ok.

weagle08, 2023-07-19:
Same issue [here](https://github.com/mavlink/MAVSDK/issues/1917). Author was unable to reproduce and we got around it by just using mavlink passthrough call to request all params. It's not as eloquent because we just use a timeout to determine when we've "should" have received all params. For us it was on startup connection so the delay in receiving the params was not an issue.

julianoes, 2023-07-21:
Thanks for the report.

> MAVSDK version: v1.4.0-418-gf7da49b4-dirty

Are you sure this is reproducible against v1.4.13? It looks like you're on a version based on something closer to the `main` branch. And note that I have recently refactored all of the param system in `main`, so this might be fixed if you use latest `main` (what is going to be v2 soon).

bwjin310, 2023-08-16:
Sorry for not replying in time. 
I did get the wrong version. Recently I have tried both `v1.4.13` and `v1.4.16`,`get_allparams()` still got error, but latest main works.
For other problems, It was found that it was caused by the telemetry plugin being declared in a timer.
- `v1.4.16`
```
[10:51:17|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
Waiting to discover system...
[10:51:17|Info ] New system on: 192.168.1.102:14552 (with sysid: 1) (udp_connection.cpp:192)
[10:51:17|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[10:51:17|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[10:51:18|Warn ] Vehicle type changed (new type: 3, old type: 0) (system_impl.cpp:225)
[10:51:18|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Discovered autopilot
Got Health:
[10:55:55|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[New Thread 0x7ffff74aa700 (LWP 55899)]
[New Thread 0x7ffff6ca9700 (LWP 55900)]
[New Thread 0x7ffff64a8700 (LWP 55901)]
Waiting to discover system...
[10:55:56|Info ] New system on: 192.168.1.102:14552 (with sysid: 1) (udp_connection.cpp:192)
[10:55:56|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[New Thread 0x7ffff5ca7700 (LWP 55902)]
[10:55:56|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[10:55:56|Warn ] Vehicle type changed (new type: 3, old type: 0) (system_impl.cpp:225)
[10:55:56|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Discovered autopilot
[10:55:56|Debug] MAVLink: info: Data link regained (system_impl.cpp:250)
Got Health:
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
Got Health:
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
Got Health:
Gyro calibration:	not ok
Accel calibration:	not ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: No associated state

Thread 4 "MAVSDK-test" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff64a8700 (LWP 55901)]
0x00007ffff766500b in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007ffff766500b in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff7644859 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff78cd8d1 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff78d937c in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff78d93e7 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff78d9699 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff78d0563 in std::__throw_future_error(int) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7b0f643 in std::_Function_handler<void (std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, mavsdk::MAVLinkParameters::ParamValue, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, mavsdk::MAVLinkParameters::ParamValue> > >), mavsdk::MAVLinkParameters::get_all_params()::{lambda(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, mavsdk::MAVLinkParameters::ParamValue, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, mavsdk::MAVLinkParameters::ParamValue> > > const&)#1}>::_M_invoke(std::_Any_data const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, mavsdk::MAVLinkParameters::ParamValue, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, mavsdk::MAVLinkParameters::ParamValue> > >&&) ()
   from /usr/lib/libmavsdk.so.1
#8  0x00007ffff7b1be27 in mavsdk::MAVLinkParameters::process_param_value(__mavlink_message const&)
    () from /usr/lib/libmavsdk.so.1
#9  0x00007ffff7b26215 in mavsdk::MAVLinkMessageHandler::process_message(__mavlink_message const&)
    () from /usr/lib/libmavsdk.so.1
#10 0x00007ffff7acbc32 in mavsdk::SystemImpl::process_mavlink_message(__mavlink_message&) ()
[... 101 lines cut ...]
Global position:	not ok
Home position:  	not ok
Got Health:
Gyro calibration:	ok
Accel calibration:	ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
Got Health:
Gyro calibration:	ok
Accel calibration:	ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
Got Health:
Gyro calibration:	ok
Accel calibration:	ok
Mag calibration:	ok
Local position: 	not ok
Global position:	not ok
Home position:  	not ok
^CPress <RETURN> to close this window...
```

julianoes, 2023-08-16:
If you could provide instructions how I can reproduce it that would be helpful.

And `main` doesn't look quite right either... :thinking:

julianoes, 2024-05-27:
I believe I fixed some of these segfaults.

closed 2024-05-27 by julianoes