mavlink/MAVSDK #2101: Disable system and component filtering

weagle08, 2023-07-19:
Is there currently a way to disable system id and component id filtering? I'm wanting to use this as a connection for a mavlink "sniffer" tool for mavlink-router and it seems that any messages not broadcast or explicitly meant for me are filtered out.

JonasVautherin, 2023-07-19:
I'd be interested in @julianoes' opinion, but the point of MAVSDK is to filter that.

If you only want to "sniff" mavlink messages, I would suggest you use a lower-level library.

weagle08, 2023-07-19:
I agree with what you say about the intent of MAVSDK, it would just be nice, and probably an easy implementation to disable this filtering via a config setting.

weagle08, 2023-07-19:
Create a release!? So browsing the source it looks like what I want has indeed already been added, it's just not in one of the tagged release versions!
[intercept_incoming_messages_async](https://github.com/mavlink/MAVSDK/blob/fadfe22e7c053945107201e2638be9a7f8b8ee5c/src/mavsdk/core/include/mavsdk/mavsdk.h#L419)