mavlink/MAVSDK #2102: Module range mismatch in "ActionImpl::set_actuator_async" function

elbarto1980, 2023-07-19:
Hi,
I think I have just found a minor bug in the ["set_actuator_async"](https://github.com/mavlink/MAVSDK/blob/fadfe22e7c053945107201e2638be9a7f8b8ee5c/src/mavsdk/plugins/action/action_impl.cpp#L516) function.
The range in the switch...case statement goes from 1 to 6 while the range of the module function (index % 6) goes from 0 to 5.
So, param6 will never be set and if (index==6), then param7 will be equal to 1 and the whole command will be ignored by PX4 [handle_message_command_both](https://github.com/PX4/PX4-Autopilot/blob/1c8ab2a0d7db2d14a6f320ebd8766b5ffaea28fa/src/modules/mavlink/mavlink_receiver.cpp#L557) function.

If I undestood correctly, at the moment PX4 only pays attention to 3 AUX outputs, namely param1, param2 and param3... so this bug should not have any effect on the whole system behavior.

julianoes, 2023-07-21:
Hi, thanks for the issue. That looks indeed a bit odd.
I can see the discussion back here: https://github.com/mavlink/MAVSDK-Proto/pull/217/files#r596590938 

What would be your suggestion how to fix it?

elbarto1980, 2023-07-21:
I think that something like this (I have not tested it yet) could solve the issue. It uses a secondary zero-based numbering.
Do you think this could work?

```
void ActionImpl::set_actuator_async(const int index, const float value,
                                    const Action::ResultCallback& callback) {
  MavlinkCommandSender::CommandLong command{};

  auto zbIndex = index - 1;  // zero-based index
  if (zbIndex >= 0) {
    command.command = MAV_CMD_DO_SET_ACTUATOR;
    command.target_component_id = _system_impl->get_autopilot_id();

    switch (zbIndex % 6) {
      case 0:
        command.params.maybe_param1 = value;
        break;
      case 1:
        command.params.maybe_param2 = value;
        break;
      case 2:
        command.params.maybe_param3 = value;
        break;
      case 3:
        command.params.maybe_param4 = value;
        break;
      case 4:
        command.params.maybe_param5 = value;
        break;
      case 5:
        command.params.maybe_param6 = value;
        break;
    }
    command.params.maybe_param7 = static_cast<float>(zbIndex) / 6.0f;

    _system_impl->send_command_async(
        command, [this, callback](MavlinkCommandSender::Result result, float) {
          command_result_callback(result, callback);
        });
  }
}
```

julianoes, 2023-08-07:
Ok, would that be according to the MAVLink spec then?

elbarto1980, 2023-08-22:
I think so. The MAVLink spec is implemented in the "command" structure, isn't it?

julianoes, 2023-08-23:
Oh, now it dawned on my :man_facepalming:! Can you make a pull request to fix it? That would be great, thanks.

elbarto1980, 2023-08-23:
Sure, I will make it ASAP (i.e. this coming weekend) 👍

closed 2024-05-27 by julianoes