mavlink/MAVSDK #2105: Adding functionality without changing the MAVSDK library

ygratz, 2023-07-25:
In order to use MAVSDK in an application, I need to follow version control requirements.
To comply with this, I can take a given MAVSDK version with a given tag.
The problem is, that sometimes I do need to make changes\adjustments in the given functionality. For example: the correction in the action.takeoff() to change mode to guided.
Is there a correct way to add such functionality to my application, while not changing the given MAVSDK tagged library I'm using?
As anyone encountered this before?
If not, is it possible to add such a gateway to allow this?

holden-zenith, 2023-07-30:
[Passthrough](https://mavsdk.mavlink.io/main/en/cpp/api_reference/classmavsdk_1_1_mavlink_passthrough.html) or intercept_outgoing_messages_async/intercept_incoming_messages_async

julianoes, 2023-08-07:
You can either fork and change but if you want to avoid that you can workaround with passthrough functionality as suggested by @holden-zenith above.

May I ask why you need to switch to Guided for Takeoff? Can you elaborate on the exact use case here?

ygratz, 2023-08-08:
Hi [julianoes](https://github.com/julianoes),
This is regarind ArduCopter.
By switching to Guided for Takeoff (maybe this is a wrong terminology): When commanding action.takeoff() - I note that I must set to guided mode in the SITL, else the takeoff() is denied.
This issue is already resolved in main branch (add check of the flight_mode).

ygratz, 2023-08-09:
Hi [julianoes](https://github.com/julianoes),
In main() branch - ActionImpl::takeoff_async() already includes an implementation which resolves this.
In case of ardupilot, it calls takeoff_async_apm() and it changes the flight mode to offboard, to allow takeoff.

JonasVautherin, 2023-08-09:
@ygratz: Do I understand correctly that it is already fixed in the `main` branch, but you want to use the latest release (which does not have this fix)?

ygratz, 2023-08-10:
Yes, that's correct

julianoes, 2023-08-11:
Ok, I don't think this is a "bugfix" that needs to go into v1.4 though. Let's get v2 out where it's fixed.

closed 2024-05-27 by julianoes