mavlink/MAVSDK #2111: ArduPilot - Possibility of using MAV_CMD_DO_SET_SERVO instead of MAV_CMD_DO_SET_ACTUATOR

trolledbypro, 2023-08-09:
Hello,

I've been messing around with the set-actuator example with my WSL2 ArduPilot SITL. I have seen that the set_actuator function is causing an unsupported message to appear in my terminal. Upon investigation of the source code: https://github.com/mavlink/MAVSDK/blob/fadfe22e7c053945107201e2638be9a7f8b8ee5c/src/mavsdk/plugins/action/action_impl.cpp#L516-L550
I see that MAV_CMD_DO_SET_ACTUATOR is being used to only set _one_ actuator to a specific value. This command is unsupported in ArduPilot (indicated by it's absence in [this list](https://ardupilot.org/dev/docs/mavlink-other-commands.html#mav-cmds))

In theory, we could use MAV_CMD_DO_SET_SERVO to accomplish what MAV_CMD_DO_SET_ACTUATOR is doing in ArduPilot.
However, I am not sure if MAV_CMD_DO_SET_ACTUATOR is supported in PX4 (https://github.com/PX4/PX4-Autopilot/pull/15281).

Would it be more preferable to change the set_actuator function to use MAV_CMD_DO_SET_SERVO or to implement a switch statement that implements MAV_CMD_DO_SET_SERVO if the AP is found to be using ArduPilot?
I could attempt to implement a case such as this: https://github.com/mavlink/MAVSDK/blob/fadfe22e7c053945107201e2638be9a7f8b8ee5c/src/mavsdk/plugins/action/action_impl.cpp#L658-L670

julianoes, 2023-08-11:
I would be ok with sending another command for ArduPilot.

JonasVautherin, 2023-08-11:
Same, I like the idea of the `if` statement, sending SET_SERVO for Ardupilot and SET_ACTUATOR for PX4:

```
if (_system_impl->autopilot() == SystemImpl::Autopilot::ArduPilot) { 
```

trolledbypro, 2023-08-11:
I implemented it in my fork: https://github.com/trolledbypro/MAVSDK/blob/93cb723a58af752895b80000179addd10a7439ea/src/mavsdk/plugins/action/action_impl.cpp#L516-L549

Looks like this is all that needs to change to get it working on ArduPilot. The two commands are equivalent due to DO_SET_ACTUATOR only being used to change one actuator at a time.

I'll submit a PR after some more testing, but this code is working with my ArduPilot SITL running set_actuator.

@julianoes How much testing/what testing do I need to do before I can submit a PR?

JonasVautherin, 2023-08-11:
> How much testing/what testing do I need to do before I can submit a PR?

If it works for you, just open a PR. It is currently not working for Ardupilot anyway, and your change should be low-risk for PX4 since you have a pretty clean `if (ardupilot)` condition :+1:.

julianoes, 2024-05-27:
Looks like that's implemented.

closed 2024-05-27 by julianoes