mavlink/MAVSDK #2116: Linker error when building with libc++

LarsLeferenz, 2023-08-14:
I'm currently trying to compile MAVSDK with llvms libc++ instead of libstdc++, due to compatibility reasons with another framework (Unreal Engine) . 

Building of most targets actually succeeds, but I can't perform an install due to linker errors in 2 components.

My CMake commands:
```sh
cmake ./MAVSDK -DCMAKE_C_COMPILER=clang \
    -DCMAKE_CXX_COMPILER=clang++ \
    -DCMAKE_CXX_STANDARD=17 \
    -DCMAKE_CXX_FLAGS="${CMAKE_CXX_FLAGS} -pthread -stdlib=libc++ -std=c++17 -lc++abi" \
    -DCMAKE_EXE_LINKER_FLAGS="${CMAKE_EXE_LINKER_FLAGS} -pthread -stdlib=libc++ -std=c++17 -lc++abi" \
    -DBUILD_SHARED_LIBS=ON \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=./ \
    -Bbuild/default 

cmake --build build/default --target install -j 6
```
Error for unit_test_runner:
```sh
[ 87%] Linking CXX executable unit_tests_runner
[ 88%] Building CXX object src/integration_tests/CMakeFiles/integration_tests_runner.dir/mission_cancellation.cpp.o
clang: warning: -lc++abi: 'linker' input unused [-Wunused-command-line-argument]
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned long, unsigned long, char const*, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::operator>>(std::__1::basic_istream<char, std::__1::char_traits<char> >&, Json::Value&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tree_node_base&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__throw_length_error(char const*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `vtable for std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_decrement(std::_Rb_tree_node_base const*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `VTT for std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::istream& std::istream::_M_extract<double>(double&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::Value::Value(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `VTT for std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_erase(unsigned long, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::Value::toStyledString() const'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ios_base::Init::~Init()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_increment(std::_Rb_tree_node_base const*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct(unsigned long, char)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_istringstream<char, std::char_traits<char>, std::allocator<char> >::basic_istringstream(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::_Ios_Openmode)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::__ostream_insert<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*, long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `vtable for std::basic_streambuf<char, std::char_traits<char> >'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ostream::operator<<(std::basic_streambuf<char, std::char_traits<char> >*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::locale::~locale()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ios_base::Init::Init()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ios_base::~ios_base()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__throw_logic_error(char const*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_increment(std::_Rb_tree_node_base*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_decrement(std::_Rb_tree_node_base*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::Value::asString() const'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find(char, unsigned long) const'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::reserve(unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::operator<<(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, Json::Value const&)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::resize(unsigned long, char)'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/unit_tests/CMakeFiles/unit_tests_runner.dir/build.make:405: src/unit_tests/unit_tests_runner] Error 1
make[1]: *** [CMakeFiles/Makefile2:1185: src/unit_tests/CMakeFiles/unit_tests_runner.dir/all] Error 2
```

Error for integration_test_runner:
```sh
[100%] Linking CXX executable integration_tests_runner
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace(unsigned long, unsigned long, char const*, unsigned long)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `Json::operator>>(std::__1::basic_istream<char, std::__1::char_traits<char> >&, Json::Value&)'
[... 20 lines cut ...]
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `vtable for std::basic_streambuf<char, std::char_traits<char> >'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ostream::operator<<(std::basic_streambuf<char, std::char_traits<char> >*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::locale::~locale()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ios_base::Init::Init()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::ios_base::~ios_base()'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::__throw_logic_error(char const*)'
/usr/bin/ld: ../mavsdk/libmavsdk.so.1.4.0: undefined reference to `std::_Rb_tree_increment(std::_Rb_tree_node_base*)'
[... cut ...]
make[2]: *** [src/integration_tests/CMakeFiles/integration_tests_runner.dir/build.make:793: src/integration_tests/integration_tests_runner] Error 1
make[1]: *** [CMakeFiles/Makefile2:1128: src/integration_tests/CMakeFiles/integration_tests_runner.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
All other components succeed.

Appearently the tests what to link against libstdc++ (going by the std::__cxx11 symbols) although I couldn't find any reference to libstdc++ in the according CMakeList.txt files.

Is there a way to force these tests to use  also use libc++ or install without them?

JonasVautherin, 2023-08-14:
We have this (unrelated but similar) `BUILD_STATIC_MAVSDK_SERVER` flag where we define `-static-libstdc++`. It appears in 3 CMakeLists.txt, including `src/mavsdk_server/test/CMakeLists.txt`.

Maybe you could try hardcoding your flags around that flag (not in the `if BUILD_STATIC_MAVSDK_SERVER` obviously, because by default this is not enabled, but put it close to it) and see if that helps?

LarsLeferenz, 2023-08-15:
Thanks for the suggestion, but I didn't really want to modify the internal CMake files of MAVSDK for CI/CD reasons. 
To still make it work, I ended up only building the `mavsdk` target and copying the library and headers manually. It now successfully links against my Unreal Engine project.

If anyone else has a similar use-case, this is the script I'm using to replace the `install` target:
https://gist.github.com/LarsLeferenz/11b05b8ff8e1255e9b6c29d21d044bfe

I can't guarantee that all plugins will link, as I only need `telemetry` for now.

closed 2023-08-15 by LarsLeferenz

JonasVautherin, 2023-08-15:
> Thanks for the suggestion, but I didn't really want to modify the internal CMake files of MAVSDK for CI/CD reasons.

I was mostly asking in an informative way: would be nice to know what it takes to make it work.