mavlink/MAVSDK #2117: Stuck on "Vehicle is getting ready to arm" in autopilot_server

souljt01, 2023-08-15:
I am using the main branch of MAVSDK, installed libmavsdk-dev_1.4.16_ubuntu20.04_amd64.deb on Ubuntu 20.04                                 
I built MAVSDK and autopilot_server:
`MAVSDK$ cmake -DCMAKE_BUILD_TYPE=Debug -Bbuild/default -H.
MAVSDK$ cmake --build build/default -j8

MAVSDK/examples/autopilot_server$ cmake -Bbuild -H.
MAVSDK/examples/autopilot_server$ cmake --build build
MAVSDK/examples/autopilot_server$ ./build/autopilot_server `
It stucks on "Vehicle is getting ready to arm" in autopilot_server
`[02:41:10|Info ] MAVSDK version: v1.4.0-511-gfadfe22e (mavsdk_impl.cpp:23)
[02:41:10|Info ] MAVSDK version: v1.4.0-511-gfadfe22e (mavsdk_impl.cpp:23)
Connected!
[02:41:10|Debug] Initializing connection to remote system... (mavsdk_impl.cpp:692)
Connected autopilot server side!
MissionRawServer created
[02:41:10|Info ] New system on: 127.0.0.1:59362 (with system ID: 1) (udp_connection.cpp:192)
[02:41:10|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:694)
[02:41:10|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[02:41:10|Debug] Discovered 1 component(s) (system_impl.cpp:548)
Discovered Autopilot from Client
[02:41:10|Debug] New: System ID: 245 Comp ID: 190 (mavsdk_impl.cpp:357)
[02:41:10|Debug] Component Ground station (190) added. (system_impl.cpp:379)
[02:41:10|Debug] Discovered 1 component(s) (system_impl.cpp:548)
[02:41:20|Debug] Setting interval for msg id: 245 interval_ms:1000 (telemetry_server_impl.cpp:36)
[02:41:21|Warn ] sending again after 0.50947 s, retries to do: 3  (512). (mavlink_command_sender.cpp:308)
[02:41:21|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:21|Warn ] sending again after 1.01834 s, retries to do: 2  (512). (mavlink_command_sender.cpp:308)
[02:41:21|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:22|Warn ] sending again after 1.52732 s, retries to do: 1  (512). (mavlink_command_sender.cpp:308)
[02:41:22|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:22|Error] Retrying failed (512) (mavlink_command_sender.cpp:336)
[02:41:25|Warn ] sending again after 0.509327 s, retries to do: 3  (512). (mavlink_command_sender.cpp:308)
[02:41:25|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:25|Warn ] sending again after 1.01831 s, retries to do: 2  (512). (mavlink_command_sender.cpp:308)
[02:41:25|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:26|Warn ] sending again after 1.52706 s, retries to do: 1  (512). (mavlink_command_sender.cpp:308)
[02:41:26|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:26|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:128)
[02:41:26|Error] Retrying failed (512) (mavlink_command_sender.cpp:336)
[02:41:27|Warn ] sending again after 0.509157 s, retries to do: 3  (512). (mavlink_command_sender.cpp:308)
[02:41:27|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:314)
[02:41:27|Warn ] sending again after 1.01795 s, retries to do: 2  (512). (mavlink_command_sender.cpp:308)
[02:41:27|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:314)
[02:41:28|Warn ] sending again after 1.52667 s, retries to do: 1  (512). (mavlink_command_sender.cpp:308)
[02:41:28|Warn ] Request was for msg ID: 280 (mavlink_command_sender.cpp:314)
[02:41:28|Error] Retrying failed (512) (mavlink_command_sender.cpp:336)
[02:41:29|Warn ] sending again after 0.508854 s, retries to do: 3  (512). (mavlink_command_sender.cpp:308)
[02:41:29|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:29|Warn ] sending again after 1.01681 s, retries to do: 2  (512). (mavlink_command_sender.cpp:308)
[02:41:29|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:30|Warn ] sending again after 1.52338 s, retries to do: 1  (512). (mavlink_command_sender.cpp:308)
[02:41:30|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:30|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:30|Error] Retrying failed (512) (mavlink_command_sender.cpp:336)
[02:41:31|Warn ] sending again, retries to do: 5 (mavlink_parameter_client.cpp:1044)
[02:41:31|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:31|Warn ] sending again, retries to do: 4 (mavlink_parameter_client.cpp:1044)
[02:41:31|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:32|Warn ] sending again, retries to do: 3 (mavlink_parameter_client.cpp:1044)
[02:41:32|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:32|Warn ] sending again, retries to do: 2 (mavlink_parameter_client.cpp:1044)
[02:41:32|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:33|Warn ] sending again after 0.507777 s, retries to do: 3  (512). (mavlink_command_sender.cpp:308)
[02:41:33|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:33|Warn ] sending again, retries to do: 1 (mavlink_parameter_client.cpp:1044)
[02:41:33|Warn ] Ignoring request_read message - param name not found: MY_PARAM (mavlink_parameter_server.cpp:369)
[02:41:33|Warn ] sending again after 1.01591 s, retries to do: 2  (512). (mavlink_command_sender.cpp:308)
[02:41:33|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:33|Error] retrying failed (mavlink_parameter_client.cpp:1061)
Creating Mission
Uploading mission...
SystemID 
[02:41:33|Debug] Receive Mission Count in Server (mission_raw_server_impl.cpp:149)
Mission uploaded.
Received Uploaded Mission!
mission_plan:
{
    mission_items: [mission_item:
{
    seq: 0
    frame: 6
    command: 16
    current: 1
    autocontinue: 1
    param1: 0.5
    param2: 1
    param3: 0
    param4: nan
    x: 473981703
    y: 85456490
    z: 10
    mission_type: 0
}, mission_item:
{
    seq: 1
    frame: 2
    command: 178
    current: 0
    autocontinue: 1
    param1: 1
    param2: 5
    param3: -1
    param4: 0
    x: 0
    y: 0
    z: nan
    mission_type: 0
}, mission_item:
{
    seq: 2
    frame: 2
    command: 204
    current: 0
    autocontinue: 1
    param1: 2
    param2: 0
    param3: 0
    param4: 1
    x: 0
    y: 0
    z: 2
    mission_type: 0
}, mission_item:
{
    seq: 3
    frame: 2
    command: 205
    current: 0
    autocontinue: 1
    param1: 20
    param2: 0
    param3: 60
    param4: nan
    x: 0
    y: 0
    z: 2
    mission_type: 0
}]
}
Current Mission Item Changed!
Current Item: mission_item:
{
    seq: 0
    frame: 6
    command: 16
    current: 1
    autocontinue: 1
    param1: 0.5
    param2: 1
    param3: 0
    param4: nan
    x: 473981703
    y: 85456490
    z: 10
    mission_type: 0
}
Current Mission Item Changed!
Current Item: mission_item:
{
    seq: 1
    frame: 2
    command: 178
    current: 0
    autocontinue: 1
    param1: 1
    param2: 5
    param3: -1
    param4: 0
    x: 0
    y: 0
    z: nan
    mission_type: 0
}
Clear All Mission!
Clear Mission Request
[02:41:34|Warn ] sending again after 1.52186 s, retries to do: 1  (512). (mavlink_command_sender.cpp:308)
[02:41:34|Warn ] Request was for msg ID: 242 (mavlink_command_sender.cpp:314)
[02:41:34|Error] Retrying failed (512) (mavlink_command_sender.cpp:336)
[02:41:53|Debug] Setting interval for msg id: 33 interval_ms:1000 (telemetry_server_impl.cpp:36)
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m
Vehicle is getting ready to arm
Altitude: 0 m
Altitude: 0 m`

julianoes, 2023-08-16:
Ok, so you implemented the autopilot server yourself? I assume you forgot to set some flag that would make it armable.

Have you set "armable"? https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action_server/include/plugins/action_server/action_server.h#L316

souljt01, 2023-08-17:
So, does that mean that autopilot_server, an example in MAVSDK, may not work properly?

souljt01, 2023-08-17:
The condition for `health_all_ok()` to be true is:
`bool TelemetryImpl::health_all_ok() const
{
    std::lock_guard<std::mutex> lock(_health_mutex);
    if (_health.is_gyrometer_calibration_ok && _health.is_accelerometer_calibration_ok &&
        _health.is_magnetometer_calibration_ok && _health.is_local_position_ok &&
        _health.is_global_position_ok && _health.is_home_position_ok) {
        return true;
    } else {
        return false;
    }
}`
And in the log I tested, it shows `is_global_position_ok: 0`.
In autopilotThread, publish_position() is executed in while loop.
`telemServer.publish_position(position, velocity, heading);`

And in the `publish_position()`, you can see the code related to global.
`mavlink_msg_global_position_int_pack(...)
add_msg_cache(MAVLINK_MSG_ID_GLOBAL_POSITION_INT, msg);`

I don't know why `is_global_position_ok` is 0.

julianoes, 2023-08-18:
> So, does that mean that autopilot_server, an example in MAVSDK, may not work properly?

Your testing will tell you. I'm sorry this is not something that is tested often.

labtoast, 2024-05-31:
The health of the GPS was not getting reevaluated so it never became good. It needed the sys_status message to be received before it would mark the GPS as good. I created a pull request to fix this: #2321

julianoes, 2024-05-31:
Thanks for fixing it @labtoast!

closed 2024-05-31 by julianoes