mavlink/MAVSDK #2119: Can't do goto_location or do_orbit with C++ SDK

SebastianRosewick, 2023-08-18:
Hello,

i am having trouble with the C++ SDK. I am using v1.4.15 with PX4 sitl and gazebo garden.

Problem Description:
I can't command the x500 quadcopter to reposition neither to do orbits. I tried using the mavsdk::Action Plugin with goto_location() and do_orbit() as well as sending a CommandInt with the MavlinkPassthrough Plugin.
This causes the quadcopter to land immediately. However, the commands are submitted successfully and no error is being logged.
I also can see the messages and ACKs being sent when debugging with the wireshark lua-plugin.
Sending the commands manually with QGC does work as well as the Python example for do_orbit.

How to reproduce:
`make px4_sitl_default gz_x500`

To isolate the problem, i added this bit of code in the takeoff_and_land example, after the system has taken off and waited for a few seconds:

` auto orbit_res = action.do_orbit(20.f, 2.f, mavsdk::Action::OrbitYawBehavior::HoldFrontToCircleCenter, 47.398036222362471, 8.5450146439425509, 2.5);
    std::cout << "Orbit_Res: " << orbit_res << std::endl;
    if(orbit_res == mavsdk::Action::Result::Success){
        sleep_for(seconds(30));
    }`

The result is that do_orbit returns SUCCESS but the drone lands immediately.
A zipped flight log is attached.

Any help would be much appreciated!
[log_17_2023-8-18-23-08-13.ulg.zip](https://github.com/mavlink/MAVSDK/files/12384017/log_17_2023-8-18-23-08-13.ulg.zip)

julianoes, 2023-08-20:
Thanks for the issue, that's a shame! Is this against PX4 and the main branch?

SebastianRosewick, 2023-08-21:
Yes, I tried it with the latest on main and also with the release/1.14 branch.

julianoes, 2023-08-23:
Alright, let me try to reproduce.

julianoes, 2023-08-23:
Ok, I tried orbit and this worked for me:

```
diff --git a/examples/takeoff_and_land/takeoff_and_land.cpp b/examples/takeoff_and_land/takeoff_and_land.cpp
index 598a3329..b4d01325 100644
--- a/examples/takeoff_and_land/takeoff_and_land.cpp
+++ b/examples/takeoff_and_land/takeoff_and_land.cpp
@@ -89,6 +89,12 @@ int main(int argc, char** argv)
     // Let it hover for a bit before landing again.
     sleep_for(seconds(10));
 
+    auto orbit_res = action.do_orbit(10, 5, Action::OrbitYawBehavior::HoldFrontToCircleCenter, 47.3977977, 8.5469410, 500);
+    std::cout << "Orbit_Res: " << orbit_res << std::endl;
+    if(orbit_res == mavsdk::Action::Result::Success){
+        sleep_for(seconds(60));
+    }
+
     std::cout << "Landing...\n";
     const Action::Result land_result = action.land();
     if (land_result != Action::Result::Success) {
```

Can you try that? I'm testing with Gazebo Classic but that shouldn't make a difference. And I'm on the PX4 branch `release/1.14`.

julianoes, 2023-08-23:
Oh, so it's the altitude! We should figure out what happens when the altitude is too low and how/whether it fails.

julianoes, 2023-08-23:
Ok, PX4 answers with "Command Denied", so I think that makes somewhat sense.

```
Arming...
Arming failed: Command Denied
```

closed 2023-08-23 by julianoes