mavlink/MAVSDK #2120: Camera plugin telemetries using deprecated subscription messages

jperez-droneup, 2023-08-22:
When subscribing to camera telemetry information, I get warnings from MAVSDK that it is ignoring the commands and I don't get the callbacks called as a result.
`[02:15:11|Warn ] Received ack for not-existing command: 522! Ignoring... (mavlink_command_sender.cpp:279)`

https://github.com/mavlink/MAVSDK/blob/84eab2d996040b6cb839f062f02d072e8723807e/src/mavsdk/plugins/camera/camera_impl.cpp#L373

https://github.com/mavlink/MAVSDK/blob/84eab2d996040b6cb839f062f02d072e8723807e/src/mavsdk/plugins/camera/camera_impl.cpp#L441

https://github.com/mavlink/MAVSDK/blob/84eab2d996040b6cb839f062f02d072e8723807e/src/mavsdk/plugins/camera/camera_impl.cpp#L452

julianoes, 2023-08-23:
Hm, can you tell me how I can reproduce the issue? What are you running this against?

jperez-droneup, 2023-08-23:
We are using MAVSDK at SHA `5d47278251276aea059602d68434547cf9e6d1ff`, but it looks like this is still the case on trunk. 

I am able to reproduce just by calling `mavsdk::Camera::subscribe_mode`, which eventually calls the `make_command_request_camera_settings` linked above. 
https://github.com/mavlink/MAVSDK/blob/84eab2d996040b6cb839f062f02d072e8723807e/src/mavsdk/plugins/camera/include/plugins/camera/camera.h#L663

According to https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_CAMERA_SETTINGS, https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION, and https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_VIDEO_STREAM_STATUS, these messages are deprecated and should be replaced with https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE. 

We are running PX4 sitl gazebo version v1.13.3.

julianoes, 2023-08-23:
Regarding replacing those messages, there is a long discussion here: https://github.com/mavlink/MAVSDK/issues/2013 It looks like we came to a way forward but haven't had time to actually implement it.

julianoes, 2023-08-23:
> We are running PX4 sitl gazebo version v1.13.3.

Which command? Is there actually a camera simulated when you run this?

jperez-droneup, 2023-08-30:
> Is there actually a camera simulated when you run this?

We use the `px4_sitl gazebo_typhoon_h480` model which has a camera configured. 

> Which command?

Not sure which command you are referring to. This?
> I am able to reproduce just by calling mavsdk::Camera::subscribe_mode, which eventually calls the make_command_request_camera_settings linked above.

julianoes, 2025-06-02:
We're now using both, new and old commands, for backwards compatibility, so I can close this.

closed 2025-06-02 by julianoes