mavlink/MAVSDK #2121: mission_raw missionitem

ShanbinZhang, 2023-08-24:
std::vector<MissionRaw::MissionItem> reference_items;
    MissionRaw::MissionItem new_raw_item_nav0;
    new_raw_item_nav.seq = 0;
    new_raw_item_nav.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav.command = MAV_CMD_NAV_TAKEOFF; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav.current = 1;
    new_raw_item_nav.autocontinue = 1;
    new_raw_item_nav.param1 = 0.0f; // Hold
    new_raw_item_nav.param2 = 0.0f; // Accept Radius
    new_raw_item_nav.param3 = 0.0f; // Pass Radius
    new_raw_item_nav.param4 = 0.0f; // Yaw
    new_raw_item_nav.x = -35.36326310;
    new_raw_item_nav.y = 149.16523800;
    new_raw_item_nav.z = 50.0f;
    new_raw_item_nav.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

    MissionRaw::MissionItem new_raw_item_nav1;
    new_raw_item_nav1.seq = 1;
    new_raw_item_nav1.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav1.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav1.current = 0;
    new_raw_item_nav1.autocontinue = 1;
    new_raw_item_nav1.param1 = 0.0f; // Hold
    new_raw_item_nav1.param2 = 0.0f; // Accept Radius
    new_raw_item_nav1.param3 = 0.0f; // Pass Radius
    new_raw_item_nav1.param4 = 0.0f; // Yaw
    new_raw_item_nav1.x = -35.36326120;
    new_raw_item_nav1.y = 149.16524010;
    new_raw_item_nav1.z = 50.0f;
    new_raw_item_nav1.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

    MissionRaw::MissionItem new_raw_item_nav2;
    new_raw_item_nav3.seq = 2;
    new_raw_item_nav3.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav3.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav3.current = 0;
    new_raw_item_nav3.autocontinue = 1;
    new_raw_item_nav3.param1 = 0.0f; // Hold
    new_raw_item_nav3.param2 = 0.0f; // Accept Radius
    new_raw_item_nav3.param3 = 0.0f; // Pass Radius
    new_raw_item_nav3.param4 = 0.0f; // Yaw
    new_raw_item_nav3.x = -35.36325940;
    new_raw_item_nav3.y = 149.16523880;
    new_raw_item_nav3.z = 50.0f;
    new_raw_item_nav3.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

    MissionRaw::MissionItem new_raw_item_nav3;
    new_raw_item_nav4.seq = 3;
    new_raw_item_nav4.frame = MAV_FRAME_GLOBAL_RELATIVE_ALT; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav4.command = MAV_CMD_NAV_WAYPOINT; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav4.current = 0;
    new_raw_item_nav4.autocontinue = 1;
    new_raw_item_nav4.param1 = 0.0f; // Hold
    new_raw_item_nav4.param2 = 0.0f; // Accept Radius
    new_raw_item_nav4.param3 = 0.0f; // Pass Radius
    new_raw_item_nav4.param4 = 0.0f; // Yaw
    new_raw_item_nav4.x = -35.36325950;
    new_raw_item_nav4.y = 149.16523720;
    new_raw_item_nav4.z = 50.0f;
    new_raw_item_nav4.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

    MissionRaw::MissionItem new_raw_item_nav4;
    new_raw_item_nav5.seq = 4;
    new_raw_item_nav5.frame = MAV_FRAME_MISSION; // MAV_FRAME_GLOBAL_RELATIVE_ALT_INT
    new_raw_item_nav5.command = MAV_CMD_NAV_RETURN_TO_LAUNCH; // MAV_CMD_NAV_WAYPOINT
    new_raw_item_nav5.current = 0;
    new_raw_item_nav5.autocontinue = 1;
    new_raw_item_nav5.param1 = 0.0f; // Hold
    new_raw_item_nav5.param2 = 0.0f; // Accept Radius
    new_raw_item_nav5.param3 = 0.0f; // Pass Radius
    new_raw_item_nav5.param4 = 0.0F; // Yaw
    new_raw_item_nav5.x = 0;
    new_raw_item_nav5.y = 0;
    new_raw_item_nav5.z = 0.0f;
    new_raw_item_nav5.mission_type = MAV_MISSION_TYPE_MISSION; // MAV_MISSION_TYPE_MISSION

    reference_items.push_back(new_raw_item_nav0);
    reference_items.push_back(new_raw_item_nav1);
    reference_items.push_back(new_raw_item_nav2);
    reference_items.push_back(new_raw_item_nav3);
    reference_items.push_back(new_raw_item_nav4);

The above is the route task uploaded by me, but the UAV did not complete the execution and was executed to new_raw_item_nav2. When the UAV started to return, it did not execute the waypoint new_raw_item_nav3. The code shows that the waypoint in new_raw_item_nav3 has been uploaded, but it was not executed. What is the reason for this? thank you!!!

julianoes, 2023-08-24:
I don't know without seeing a logfile. If you're using PX4, you can upload your logfile of the flight to logs.px4.io and then share the link here.

ShanbinZhang, 2023-08-24:
> I don't know without seeing a logfile. If you're using PX4, you can upload your logfile of the flight to logs.px4.io and then share the link here.

used APM,That's a strange question

julianoes, 2023-08-25:
Ok, I'm not sure how to debug it then. Surely, you have some sort of log that tells you what was going on? Could it be that the waypoints are very close together and it "reached" both very rapidly?

julianoes, 2025-06-02:
No more responses, closing.