mavlink/MAVSDK #2128: set flight mode function

alireza787b, 2023-09-06:
Hi
I was just tryin to build a new function to set the flight mode command in mavsdk. I created the proto s and were just implementing the cpp when I found that there is already a function there called set_flight_mode_async that does it. is that correct? how can I change flight mode from python to sth like hold?

julianoes, 2023-09-06:
For PX4, to go into Hold, you can do `mission->pause`.

For proper support of flight modes, I'd like to use the the common flight modes which is a new MAVLink spec allowing standard and custom modes which would be (eventually) compatible with both PX4 and ArduPilot: https://github.com/mavlink/mavlink/blob/fe14d798662f687407b03e383d2a9841a8590392/message_definitions/v1.0/development.xml#L473-L502.

alireza787b, 2023-09-07:
what if I am not already in a mission mode? maybe in offboard switching to hold (maybe stopping offboard would do that) but generally eg. on the ground I want to change flight mode to position hold. or stabilized 
I was trying to so sth like this :
Action::Result ActionImpl::SetFlightMode(const std::string& flight_mode)
{
    // Initialize variables for the MAVLink command
    uint8_t base_mode = 0;
    uint8_t main_mode = 0;
    uint8_t sub_mode = 0;

    // Convert flight_mode string to base_mode, main_mode, and sub_mode
    if (flight_mode == "Manual") {
        base_mode = 217;
        main_mode = 1;
        sub_mode = 0;
    } else if (flight_mode == "Stabilized") {
        base_mode = 209;
        main_mode = 7;
        sub_mode = 0;
    } else if (flight_mode == "Acro") {
        base_mode = 209;
        main_mode = 5;
        sub_mode = 0;
    } else if (flight_mode == "Rattitude") {
        base_mode = 193;
        main_mode = 8;
        sub_mode = 0;
    } else if (flight_mode == "Altitude") {
        base_mode = 193;
        main_mode = 2;
        sub_mode = 0;
    } else if (flight_mode == "Offboard") {
        base_mode = 209;
        main_mode = 6;
        sub_mode = 0;
    } else if (flight_mode == "Position") {
        base_mode = 209;
        main_mode = 3;
        sub_mode = 0;
    } else if (flight_mode == "Hold") {
        base_mode = 217;
        main_mode = 4;
        sub_mode = 3;
    } else {
        // If the mode is not recognized, return an error
        return Action::Result::UNKNOWN_ERROR;
    }

    // Prepare the MAVLink message
    mavlink_message_t msg;
    mavlink_msg_command_long_pack(
        _parent->get_own_system_id(),
        _parent->get_own_component_id(),
        &msg,
        _parent->get_target_system_id(),
        _parent->get_target_component_id(),
        MAV_CMD_DO_SET_MODE,  // command id
        0,                    // confirmation
        base_mode,             // base_mode
        main_mode,             // main_mode
        sub_mode,              // sub_mode
        0, 0, 0, 0             // unused parameters
    );

    // Send the MAVLink message and receive the result
    auto send_result = _parent->send_message(msg);

    // Convert the MavlinkCommandSender::Result to Action::Result and return
    return action_result_from_command_result(send_result);
}

I go this mode codes from forums . when I try to implement this I notice there is already  a set_flight_mode in the codes. just wanted to make sure if I am on correct path or doing sth wrong?

`
void ActionImpl::hold_async(const Action::ResultCallback& callback) const
{
    _system_impl->set_flight_mode_async(
        FlightMode::Hold, [this, callback](MavlinkCommandSender::Result result, float) {
            command_result_callback(result, callback);
        });
}
`

julianoes, 2023-09-07:
> what if I am not already in a mission mode?

It doesn't matter. Check what it does:
https://github.com/mavlink/MAVSDK/blob/d42e9f2169a07cc664282b719e609103e0adbf27/src/mavsdk/plugins/mission/mission_impl.cpp#L906-L912

Yes, you should be able to build on top of the `set_flight_mode`.

acofis, 2023-09-11:
Is there any existing code for setting other modes?

I am starting work with ardupilot drones and need them to be set to "guided" before they can get a takeoff command, and must return from "land" to "guided" to fly again.

I am considering contributing once I have looked over the methodology for doing so but wanted to be sure this is not already in place

JonasVautherin, 2023-09-11:
> I am starting work with ardupilot drones and need them to be set to "guided" before they can get a takeoff command

You should not have to do that, MAVSDK should set the mode correctly when you `drone.action.takeoff()`. However, maybe that's not yet in the latest release v1.4.x.

For Ardupilot you should use the latest `main` branch [until v2 is out](https://github.com/mavlink/MAVSDK/issues/2002), I would say.

acofis, 2023-09-11:
> 

I was using the latest 1.4 built server exe for windows. I will attempt to buld main and run the same command to see how this goes.

acofis, 2023-09-12:
> You should not have to do that, MAVSDK should set the mode correctly when you `drone.action.takeoff()`. However, maybe that's not yet in the latest release v1.4.x.
>
> For Ardupilot you should use the [...]

I am attempting to build a mavDSK-server.exe from main and I have 2 problems:

1. The [instructions ](https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html)do not mention windows.
2. The lines provided for Linux and mac (which are identical as far as I can tell) do not work

Am I just missing something in the docs or is windows missing specifically? Should I make a new ticket to discuss this?

Lastly are there nightly builds of the binaries that I could use to get the latest snapshot of main and avoid this build process?