mavlink/MAVSDK #2130: mission-variable as member of C++-Class

expeditionengineer, 2023-09-07:
Hi Guys,

I have a question, which is more C++ related than MAVSDK related, but maybe you will still help me.
I have a ROS2 node-class in which a method is executed based on a timer with a specific frequency. I now want to command a mission, using the MAVSDK-Mission plugin inside this callback-method and want to start the mission in one cycle and track the progress of the mission in every cycle of the callback-method. For that i checked the fly_mission.cpp in the MAVSDK 1.4.10 examples-code and adapted it. For that i need to define the `mission`-variable as a member variable of the ROS2-Node class, but i struggle with it, since its type is deduced with `auto`.  I tried to set the type of mission with something like this `std::shared_ptr<mavsdk::Mission> mission;` but i cant assign it to the instantiation of the Mission-class without compiler error. Maybe you can tell me, if it is possible to save the mission-variable as a class memeber variable and tell me the type, that would help me alot. Sorry for the nooby-question :D   
(From fly_mission.cpp example):
```
    auto action = Action{system};
    auto mission = Mission{system};
    auto telemetry = Telemetry{system};
```
Greetings, Tobias

julianoes, 2023-09-07:
Hi Tobias,

I think we do something similar here:
https://github.com/PX4/PX4-Autopilot/blob/ad1d72df6fc140999c3ca31e6b138fd7d1498728/test/mavsdk_tests/autopilot_tester.h#L269-L278

And then instantiate the plugins here:
https://github.com/PX4/PX4-Autopilot/blob/ad1d72df6fc140999c3ca31e6b138fd7d1498728/test/mavsdk_tests/autopilot_tester.cpp#L68-L78

Does that work?

expeditionengineer, 2023-09-12:
I works perfectly, thanks alot!

closed 2023-09-12 by expeditionengineer