mavlink/MAVSDK #2137: Action->set_maximum_speed doesn't work with ArduPilot?

paulholmstedt, 2023-09-11:
I can't get MAVSDK Action->set_maximum_speed to work with ArduPilot. Looking at the MAVSDK code, it is using `MAX_SPEED_PARAM `(which is actually `MPC_XY_CRUISE`) to control the maximum speed. But for ArduPilot the parameter controlling the maximum speed in at least Auto (waypoint) and Guided (with `MAV_CMD_DO_REPOSITION`) is `WPNAV_SPEED`. Also unless you set `RTL_SPEED`, the `WPNAV_SPEED` parameter will be used for that as well. Also tried sending the MAV_CMD_DO_CHANGE_SPEED command with passthrough but I couldn't get ArduPilot to react to my commands in Guide mode. So right now I use MAVSDK Param->set_param_float with `WPNAV_SPEED` as argument every time I want to change the speed in Auto and Guided flight modes (after calling `goto_location_async`) I'm not an ArduPilot expert so I might not got it 100% correct :-)

julianoes, 2023-09-12:
I think we should actually use the command, for both PX4 and ArduPilot:

https://mavlink.io/en/messages/common.html#MAV_CMD_DO_CHANGE_SPEED

paulholmstedt, 2023-09-12:
Yes, when I have tested both PX4 and ArduPilot responds to the `MAV_CMD_DO_CHANGE_SPEED`.

PR #2435 (2024-10-30): action: remove max speed settings
This API is confusing as it only works for PX4 quads but not PX4 fixedwing or anything ArduPilot based.

The reason is that this just sets a PX4 parameter that is not standardized via MAVLink.

Therefore, I suggest to remove the API and instead recommend to use the param plugin to set the specific params specifically.

Depends on https://github.com/mavlink/MAVSDK-Proto/pull/355.

Closes #2137.
closed 2024-10-30 by julianoes