mavlink/MAVSDK #2138: motor didnt spin when arm and drone not lift off

naorwaiss, 2023-09-11:
hi, i use pixhwak cube black with arduino 
i am try to arm the motor and lift off the drone to 2 mater 
the drone move to arm but the motor didnt spine until i put some pwm from the rc 
if i try to gave some pwm the motor didnt speed fast enogh to lidt the drone 
i am alittle new at the mavsdk... thanks for the help 

this is the script :
import asyncio
from mavsdk import System

async def connect_to_drone():
    # Connect to the drone using the serial port
    drone = System()
    await drone.connect(system_address="serial:///dev/ttyTHS1")

    # Check if the drone is connected
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Drone is connected!")
            return drone  # Return the connected drone object

    print("Failed to connect to the drone.")
    return None  # Return None if connection fails

async def arm_drone(drone):
    print("Arming the drone...")
    try:
        await drone.action.arm()
        print("Drone is armed!")
    except Exception as e:
        print(f"Error arming the drone: {e}")

async def takeoff(drone):
    print("Taking off...")
    try:
        await drone.action.takeoff()
        print("Drone is ascending...")

        async for is_in_air in drone.telemetry.in_air():
            if is_in_air:
                print("Drone is in the air!")
                break

    except Exception as e:
        print(f"Error taking off: {e}")

async def main():
    drone = await connect_to_drone()
    if drone:
        await arm_drone(drone)  # Arm the drone if connected
        await takeoff(drone)  # Perform the takeoff
        # You can perform other tasks with the connected and armed drone here
        # For example, you can set waypoints, perform maneuvers, etc.

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())

julianoes, 2023-09-12:
You have to have your drone flying properly with just RC before you use it using MAVSDK. Are you using PX4 or ArduPilot?

naorwaiss, 2023-09-12:
the drone flight already.. do all the check 
I am using ardupilot 
ummm I didn't use GPS if it is neccecery ...

naorwaiss, 2023-09-15:
hi. try with gps it seems that the drone didnt change the motors spine to try accesses to his altitude

julianoes, 2023-09-18:
I'm not too familiar with ArduPilot, and you have not supplied a logfile, so I don't know how to help you.

naorwaiss, 2023-09-19:
today i try again 

/home/drone/.virtualenvs/drone-camera/bin/python /tmp/pycharm_project_812/mavSDK main.py 
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...

this is the problem i deal with

the script is :

#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/ttyTHS1")

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(10)

    print("-- Landing")
    await drone.action.land()

    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())

if i delet the section of the poston the drone arm and the motor start spin at low speed..
this is more help ?

naorwaiss, 2023-09-19:
/home/drone/.virtualenvs/drone-camera/bin/python /tmp/pycharm_project_812/mavSDK main.py 
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Taking off
Traceback (most recent call last):
  File "/tmp/pycharm_project_812/mavSDK main.py", line 50, in <module>
    asyncio.run(run())
  File "/usr/local/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/tmp/pycharm_project_812/mavSDK main.py", line 30, in run
    await drone.action.takeoff()
  File "/home/drone/.local/lib/python3.9/site-packages/mavsdk/action.py", line 368, in takeoff
    raise ActionError(result, "takeoff()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()
exception calling callback for <Future at 0x7f881c3340 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/drone/.local/lib/python3.9/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/drone/.local/lib/python3.9/site-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/home/drone/.local/lib/python3.9/site-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/concurrent/futures/_base.py", line 330, in _invoke_callbacks
    callback(self)
  File "/usr/local/lib/python3.9/asyncio/futures.py", line 398, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 796, in call_soon_threadsafe
    self._check_closed()
  File "/usr/local/lib/python3.9/asyncio/base_events.py", line 515, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed

and this is the problem after i get position

closed 2024-05-27 by julianoes