mavlink/MAVSDK #2139: Cannot build on windows following documented build process.

acofis, 2023-09-12:
> > I am starting work with ardupilot drones and need them to be set to "guided" before they can get a takeoff command
> 
> You should not have to do that, MAVSDK should set the mode correctly when you `drone.action.takeoff()`. However, maybe that's not yet in the latest release v1.4.x.
> 
> For Ardupilot you should use the latest `main` branch [until v2 is out](https://github.com/mavlink/MAVSDK/issues/2002), I would say.

I am attempting to build a mavDSK-server.exe from main and I have 2 problems:

1. The [instructions ](https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html)do not mention windows.
2. The lines provided for Linux and mac (which are identical as far as I can tell) do not work

```
 No update step for 'openssl'
  No patch step for 'openssl'
  Performing configure step for 'openssl'
  Can't locate Locale/Maketext/Simple.pm in @INC (you may need to install the Locale::Maketext::Simple module) (@INC contains: /c/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/util/perl /usr/lib/perl5/site_perl /usr/share/perl5/site_perl /usr/lib/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib/perl5/core_perl /usr/share/perl5/core_perl /c/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/external/perl/Text-Template-1.56/lib) at /usr/share/perl5/core_perl/Params/Check.pm line 6.
  BEGIN failed--compilation aborted at /usr/share/perl5/core_perl/Params/Check.pm line 6.
  Compilation failed in require at /usr/share/perl5/core_perl/IPC/Cmd.pm line 59.
  BEGIN failed--compilation aborted at /usr/share/perl5/core_perl/IPC/Cmd.pm line 59.
  Compilation failed in require at /c/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/util/perl/OpenSSL/config.pm line 19.
  BEGIN failed--compilation aborted at /c/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/util/perl/OpenSSL/config.pm line 19.
  Compilation failed in require at C:/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/Configure line 23.
  BEGIN failed--compilation aborted at C:/MAVSDK/MAVSDK/build/default/third_party/openssl/openssl/src/openssl/Configure line 23.
C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(249,5): error MSB8066: Custom build for 'C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-mkdir.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-download.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-update.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-patch.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-configure.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-build.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\9fd51fc370f4e1eee9ede0f03f96e7cc\openssl-install.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\91fe7f23fb265bc3325a0f7fdda7f7e7\openssl-complete.rule;C:\MAVSDK\MAVSDK\build\default\third_party\openssl\CMakeFiles\166370b8cb89e2a7e4ed27f6cd2c8dff\openssl.rule;C:\MAVSDK\MAVSDK\third_party\openssl\CMakeLists.txt' exited with code 2. [C:\MAVSDK\MAVSDK\build\default\third_party\openssl\openssl.vcxproj]
CMake Error at third_party/cmake/build_target.cmake:51 (message):
  C:/MAVSDK/MAVSDK/build/default/third_party/openssl failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:22 (build_target)

-- Configuring incomplete, errors occurred!
```

Am I just missing something in the docs or is windows missing specifically? Should I make a new ticket to discuss this?

Lastly are there nightly builds of the binaries that I could use to get the latest snapshot of main and avoid this build process?

_Originally posted by @acofis in https://github.com/mavlink/MAVSDK/issues/2128#issuecomment-1716301255_

JonasVautherin, 2023-09-13:
If you want to build in command line with visual studio, the CI pipeline may help, [see here](https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L570).

Otherwise as you figured it's "just" a CMake project, so the commands should be similar to Linux. You'd have to understand why openssl fails to build though. You could try to build it manually first, to see if that works.

julianoes, 2023-09-13:
I will try to reproduce but I'm usually working on Linux and not Windows.

To push this up my todo list you can always consider [sponsoring me](https://github.com/sponsors/julianoes).

acofis, 2023-09-14:
Thank you for looking into it!

I think I figured it out. It was an issue of what I was launching the command from.

Firstly I was unintentionally running it from a git bash terminal and it did not seem to like that and had a permission issue.

Then I tried it in powershell and it did not like perl (I have strawberry perl installed for other work and it could not seem to find it)

Then I ran it in plain old cmd and it all worked fine... not sure the reason but that was the fix.

julianoes, 2023-09-14:
Ok, glad you got it to build @acofis and thanks for the follow up.

closed 2023-09-14 by julianoes