mavlink/MAVSDK #2142: Adding telemetry logs

KolesnikovArt, 2023-09-18:
I think it would be usefull to add telemetry logs. They contain much more information about the flight. I'd love to add this feature if it's appropriate and someone can tell me how best to do this without slowing down the flow of subscriprions.

Telemetry logs contain each package and unix time for it. You can read more about logs here:
https://ardupilot.org/planner/docs/mission-planner-telemetry-logs.html

julianoes, 2023-09-21:
I think that's called statustext. Try https://github.com/mavlink/MAVSDK/blob/v1.4/src/mavsdk/plugins/telemetry/include/plugins/telemetry/telemetry.h#L1314-L1317.

KolesnikovArt, 2023-09-25:
I could be wrong, but statustext has nothing to do with tlog. This file format is binary and consists of raw messages and times and can be used to view the entire flight. Btw, thanks for your work :)

JonasVautherin, 2023-09-25:
So you want MAVSDK to save, the tlog format, all the mavlink messages it receives?

Could that be done in a separate process, such that it could be done in a proxy or in a node connected to e.g. mavlink-router? Not sure if MAVSDK itself should be doing that.

julianoes, 2023-09-26:
Oh, sorry I misunderstood what you meant @KolesnikovArt.

Do you have a link to the format of tlogs? I think a method to save or replay tlogs would be good, yes.

KolesnikovArt, 2023-09-26:
No, i couldn't find the link. But the format itself is quite simple. Here is an example of the code i wrote. It's Qt, but the idea is clear. But I can't find any other way other than subscribing to all messages, which is terribly wrong and slows down the queue a lot. I think, @JonasVautherin is right, it should be done in a separate thread or processs

https://github.com/KolesnikovArt/TLog

KolesnikovArt, 2023-09-26:
Oh, thanks. It makes everything much easier. If you are interested in this feature, I would be happy to help you implement it. If not, I can close the issue

closed 2023-09-27 by KolesnikovArt

JonasVautherin, 2023-09-27:
If you write code that generates a `.tlog`, feel free to share it, I would be interested :+1:. I am not a huge fan of extending MAVSDK with more and more MAVLink debugging features, because IMO that can be done separately. If all the QGC debugging tools had been made separately, for instance, they could be reused without QGC.

Ryanf55, 2026-04-16:
I will take on implementation of this. It would be a very useful feature.
Both the autopilot, Mission Planner, and QGC all generate tlogs.

Can the issue please be re-opened because it's not solved?

Ryanf55, 2026-04-20:
If he isn't, I am happy to get to it this week. 

One thing I need to figure out is how tlog sorts out which interface each message was on. We specially need that info to determine whether our mavlink router config is correct, but interface isn't part of the tlog spec

bansiesta, 2026-04-20:
Sure, I'll take a look at putting together an example. The main challenge is that MavlinkDirect exposes parsed messages (JSON fields) rather than raw wire bytes, and tlog format stores raw MAVLink packets. I'll need to re-encode via libmav or find another approach to get the raw bytes. Will report back once I have something working.

julianoes, 2026-04-20:
@Ryanf55 sorry, I probably should have made that more explicit and obvious. It's an experiment where he helps me maintain MAVSDK.

julianoes, 2026-04-24:
Alright @bansiesta can make a PR, separate from the example.