mavlink/MAVSDK #2144: Cannot subscribe to/from messages using MavlinkPassthrough

codymarquardt, 2023-09-21:
I am running Ardusub on a blueROV and we have a Jetson Nano onboard that is communicating over serial cable with the onboard Pixhawk. My jetson nano is running Ubuntu 18.04 and I built the main branch of MAVSDK from source two days ago. I installed system-wide (so /usr/local) and I am using Qt to run a program that pulls state information from the vehicle and publishes thruster commands.

I can publish commands just fine using MavlinkPassthrough, but I cannot utilize the subscribe methods to asynchronously update state information. 

If I create a method as follows 

`void BlueROV_Controls::set_message_callback(MavlinkPassthrough &mavlink_pass_){

    MavlinkPassthrough::MessageCallback m_callback;

    m_callback = [](const mavlink_message_t &msg_)
    {
        const mavlink_message_t* msg = &msg_;
        auto pitch = mavlink_msg_nav_controller_output_get_nav_pitch(msg);
        qDebug() << QString::number(pitch) << "\n";
    };

    uint16_t value = 62;
    auto const result = mavlink_pass_.subscribe_message(value, m_callback);
}`

And call that method from main.cpp, I get the following error

error: undefined reference to `mavsdk::MavlinkPassthrough::subscribe_message(unsigned short, std::function<void (__mavlink_message const&)> const&)'

It should not be a linking issue, I can run MavlinkPassthrough::send_message() and MavlinkPassthrough::send_command_long(), for example, with no issue.

I checked the header files for MavlinkPassthrough and the API is matching the syntax I am using the best of my knowledge. 

Any help on this would be greatly appreciated!

JonasVautherin, 2023-09-21:
Which version of MAVSDK are you using?

codymarquardt, 2023-09-21:
can you elaborate on that? I built the source code off of main, i did not use a specific release

JonasVautherin, 2023-09-21:
Oh right, sorry I had missed that in your issue. So [`subscribe_message` does exist](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough_impl.h#L38) indeed, I just wanted to verify that.

codymarquardt, 2023-09-21:
Yes, it's at line 192 of /usr/local/include/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h

Not sure if this helps, but I also cannot use certain functions within telemetry. For example, I can call telemetry.set_rate_position(10.0);

but not 
telemetry.set_rate_attitude_euler(10.0); 
telemetry.set_rate_attitude_quaternion(10.0);

all of which exist in the API

codymarquardt, 2023-09-22:
Any further thoughts on this?

JonasVautherin, 2023-09-22:
Feels like something is wrong in your C++ code. The function exists in the header file, so it seems like you are not calling it correctly.

codymarquardt, 2023-09-22:
Any thoughts as to what? I'm following the syntax used in the API so I'm not sure what the issue could be

JonasVautherin, 2023-09-23:
What error do you get when you call `telemetry.set_rate_attitude_euler(10.0);`?

codymarquardt, 2023-09-23:
I get an undefined reference error. 

I just built the code and ran it on a ubuntu 18.04 amd machine and it builds fine. I am wondering if the arm architecture of the Jetson nano could cause any issue.

I also may try pulling the v1.4 release instead of main since I see the API changed

julianoes, 2023-10-05:
> I get an undefined reference error.

What exactly is the error?

julianoes, 2024-05-27:
No more follow up, closing.

closed 2024-05-27 by julianoes