mavlink/MAVSDK #2147: ulpoad mission to piwhaxk 4 mini via serial in HITL

expeditionengineer, 2023-09-27:
Hi Guys, 

I have a issue with uploading a mission via serial link from an raspberry Pi 4B to pixhawk 4 mini running PX4 1.14 in HITL-mode with JMavSim. In SITL everything is working fine btw :D

[09:22:56|Info ] MAVSDK version: v1.4.17 (mavsdk_impl.cpp:20)
Waiting to discover system...
[09:22:57|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[09:22:57|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[09:22:57|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[09:22:57|Debug] Discovered 1 component(s) (system_impl.cpp:578)
Uploading mission...
[09:22:58|Warn ] sending again after 0.507569 s, retries to do: 3  (512). (mavlink_command_sender.cpp:287)
[09:22:58|Warn ] sending again after 0.508156 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[09:22:58|Warn ] sending again, retries to do: 3  (CAL_GYRO0_ID). (mavlink_parameters.cpp:1291)
[09:22:58|Debug] Falling back to gimbal protocol v1 (mission_impl.cpp:124)
[09:22:59|Warn ] sending again after 1.01415 s, retries to do: 2  (520). (mavlink_command_sender.cpp:287)
[09:22:59|Warn ] sending again after 1.01454 s, retries to do: 2  (512). (mavlink_command_sender.cpp:287)
[09:22:59|Warn ] sending again, retries to do: 2  (CAL_GYRO0_ID). (mavlink_parameters.cpp:1291)
[09:22:59|Warn ] command denied (512). (mavlink_command_sender.cpp:191)
[09:22:59|Warn ] sending again, retries to do: 3  (CAL_ACC0_ID). (mavlink_parameters.cpp:1291)
[09:22:59|Warn ] sending again after 1.52004 s, retries to do: 1  (520). (mavlink_command_sender.cpp:287)
[09:23:00|Warn ] sending again, retries to do: 2  (CAL_ACC0_ID). (mavlink_parameters.cpp:1291)
[09:23:00|Warn ] sending again, retries to do: 1  (CAL_ACC0_ID). (mavlink_parameters.cpp:1291)
[09:23:01|Error] Error: Retrying failed get param busy timeout: CAL_ACC0_ID (mavlink_parameters.cpp:1320)
[09:23:01|Error] Error: Param for accel cal failed. (telemetry_impl.cpp:1580)
[09:23:01|Warn ] sending again, retries to do: 3  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[09:23:01|Warn ] timeout: retries exceeded (mavlink_mission_transfer.cpp:535)

I Guess it complaince about no IMU-data present? Did someone had a similar issue and found a solution? I need to test if the mission-upload is working beore go to real flight Test.  Thanks for help :)

Greetings, Tobias

JonasVautherin, 2023-09-27:
> I Guess it complaince about no IMU-data present?

No it complains about failing to talk to the drone. Seems like MAVSDK receives MAVLink from the drone, but the drone does not receive MAVLink from MAVSDK.

expeditionengineer, 2023-09-27:
Thanks Jonas. So the serial connection is the problem. I will debug it.

closed 2023-09-27 by expeditionengineer