mavlink/MAVSDK #2149: Python: drone.connect() to remote SITL/PX4 hangs

mfattori, 2023-10-06:
Hi There,

I have an instance of SITL/PX4 running on an Ubuntu (20.04) system (192.168.2.116). When the instance starts it displays:

INFO [commander] LED: open /dev/led0 failed (22)
WARN [health_and_arming_checks] Preflight Fail: ekf2 missing data
INFO [navigator] Mission #2 loaded, 7 WPs
INFO [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
…

I am trying to connect to the instance using MAVSDK from a different Ubuntu (20.04) system at 192.168.2.120
e.g. start apython…

from mavsdk import System
drone = System()
await drone.connect(system_address=“udp://192.168.2.116:18570”)

I can see the SITL instance recognizes the connection as it prints:

INFO [mavlink] partner IP: 192.168.2.120

but the ‘await’ connection line just hangs and I can not get beyond this line. I started the SITL/PX4 instance with 

'make px4_sitl_default jmavsim 

and it appears to be working well.

I’m using Python 3.8. Is there some way I can debug why it's hanging?

Thanks

JonasVautherin, 2023-10-06:
On your first system (192.168.2.116), can you try running SITL with this instead (just for the test):

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.13.3 192.168.2.120
```

And on the second system (192.168.2.120) run MAVSDK with:

```python
from mavsdk import System
drone = System()
await drone.connect(system_address=“udp://:14540”)
```

Does this work?

mfattori, 2023-10-06:
Hi Jonas,

Thank you for your reply.

Yes, doing it this way works properly. I can arm, takeoff, etc.

mfattori, 2023-10-07:
More information ...

1. If I start the container instance on 192.168.2.116... I can connect to it and run commands from ...120 and they work, as I mentioned.
2. Then I shut down the container instance and start the 'normal' instance with 'px4_sitl_default jmavsim'
3. The connection to sitl is maintained and running mavsdk commands on ...120 work with the new instance.  
4. But if I don't begin with the container instance I still cannot connect

This is a bit weird (to me)

julianoes, 2023-10-07:
> 4. But if I don't begin with the container instance I still cannot connect

How are you doing that? How can I reproduce that?

Without docker I would use: `await drone.connect(system_address=“udp://:14540”)`

JonasVautherin, 2023-10-07:
> Then I shut down the container instance and start the 'normal' instance with 'px4_sitl_default jmavsim'

How do you expect jmavsim on 192.168.2.116 will reach MAVSDK on 192.168.2.120? Does jmavsim broadcast MAVLink on the subnet? I would expect it doesn't :thinking:

Can you try running mavlink-router next to your jmavsim, with mavlink-router listening on 14550/udp and sending to 192.168.2.120:14540/udp?

mfattori, 2023-10-08:
OK, thanks, got it. 

It works properly with mavlink-routerd running and I can connect to SITL and run commands.

Thank you!

closed 2023-10-08 by julianoes