mavlink/MAVSDK #2156: I am interfacing storm32 gimbal and having some issues during compilation

shivamydv2001, 2023-10-16:
In file included from /usr/include/mavsdk/plugins/telemetry/telemetry.h:16,
                 from gimbal.cpp:12:
/usr/include/mavsdk/plugins/telemetry/plugin_base.h:18:30: error: redefinition of ‘struct mavsdk::overloaded<Ts>’
   18 | template<class... Ts> struct overloaded : Ts... {
      |                              ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/gimbal/gimbal.h:16,
                 from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/plugin_base.h:18:30: note: previous definition of ‘struct mavsdk::overloaded<Ts>’
   18 | template<class... Ts> struct overloaded : Ts... {
      |                              ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/telemetry/telemetry.h:16,
                 from gimbal.cpp:12:
/usr/include/mavsdk/plugins/telemetry/plugin_base.h:30:7: error: redefinition of ‘class mavsdk::PluginBase’
   30 | class PluginBase {
      |       ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/gimbal/gimbal.h:16,
                 from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/plugin_base.h:30:7: note: previous definition of ‘class mavsdk::PluginBase’
   30 | class PluginBase {
      |       ^~~~~~~~~~
gimbal.cpp: In function ‘int main(int, char**)’:
gimbal.cpp:47:26: error: ‘class mavsdk::Mavsdk’ has no member named ‘first_autopilot’
   47 |     auto system = mavsdk.first_autopilot(3.0);

these are all the issues , please help me to solve all these problems . I am very beginner in this project.

julianoes, 2023-10-16:
Have you changed anything?

Using a storm32 gimbal would probably require changes on the autopilot side to deal with, not in mavsdk.

shivamydv2001, 2023-10-16:
Thanks for your kind reply

I am just trying to compile and getting these errors.

shivamydv2001, 2023-10-17:
> Have you changed anything?
> 
> Using a storm32 gimbal would probably require changes on the autopilot side to deal with, not in mavsdk.

I did not change anything.

julianoes, 2023-10-17:
Ok, so the question has nothing to do with storm32.

What are your build commands?

julianoes, 2023-10-17:
Ok, so the question has nothing to do with storm32.

What are your build commands?

What's the full output from a clean build?

shivamydv2001, 2023-10-18:
connection with hardware is compulsory?

shivamydv2001, 2023-10-18:
I installed MAVSDK properly agian and I got same error 

here is the full output /MAVSDK/examples/gimbal$ g++ gimbal.cpp -o gimbal
In file included from /usr/include/mavsdk/plugins/gimbal/gimbal.h:16,
                 from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/plugin_base.h:25:61: error: expected constructor, destructor, or type conversion before ‘;’ token
   25 | template<class... Ts> overloaded(Ts...) -> overloaded<Ts...>;
      |                                                             ^
In file included from /usr/include/mavsdk/plugins/telemetry/telemetry.h:16,
                 from gimbal.cpp:11:
/usr/include/mavsdk/plugins/telemetry/plugin_base.h:18:30: error: redefinition of ‘struct mavsdk::overloaded<Ts>’
   18 | template<class... Ts> struct overloaded : Ts... {
      |                              ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/gimbal/gimbal.h:16,
                 from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/plugin_base.h:18:30: note: previous definition of ‘struct mavsdk::overloaded<Ts>’
   18 | template<class... Ts> struct overloaded : Ts... {
      |                              ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/telemetry/telemetry.h:16,
                 from gimbal.cpp:11:
/usr/include/mavsdk/plugins/telemetry/plugin_base.h:31:7: error: redefinition of ‘class mavsdk::PluginBase’
   31 | class PluginBase {
      |       ^~~~~~~~~~
In file included from /usr/include/mavsdk/plugins/gimbal/gimbal.h:16,
                 from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/plugin_base.h:30:7: note: previous definition of ‘class mavsdk::PluginBase’
   30 | class PluginBase {
      |       ^~~~~~~~~~
gimbal.cpp: In function ‘int main(int, char**)’:
gimbal.cpp:46:26: error: ‘class mavsdk::Mavsdk’ has no member named ‘first_autopilot’
   46 |     auto system = mavsdk.first_autopilot(3.0);
      |                          ^~~~~~~~~~~~~~~
gimbal.cpp:53:46: error: no matching function for call to ‘mavsdk::Telemetry::Telemetry(<brace-enclosed initializer list>)’
   53 |     auto telemetry = Telemetry{system.value()};
      |                                              ^
In file included from gimbal.cpp:11:
/usr/include/mavsdk/plugins/telemetry/telemetry.h:1700:5: note: candidate: ‘mavsdk::Telemetry::Telemetry(const mavsdk::Telemetry&)’
 1700 |     Telemetry(const Telemetry& other);
      |     ^~~~~~~~~
/usr/include/mavsdk/plugins/telemetry/telemetry.h:1700:5: note:   conversion of argument 1 would be ill-formed:
/usr/include/mavsdk/plugins/telemetry/telemetry.h:53:14: note: candidate: ‘mavsdk::Telemetry::Telemetry(std::shared_ptr<mavsdk::System>)’
   53 |     explicit Telemetry(std::shared_ptr<System> system); // new
      |              ^~~~~~~~~
/usr/include/mavsdk/plugins/telemetry/telemetry.h:53:14: note:   conversion of argument 1 would be ill-formed:
/usr/include/mavsdk/plugins/telemetry/telemetry.h:40:14: note: candidate: ‘mavsdk::Telemetry::Telemetry(mavsdk::System&)’
   40 |     explicit Telemetry(System& system); // deprecated
      |              ^~~~~~~~~
/usr/include/mavsdk/plugins/telemetry/telemetry.h:40:14: note:   conversion of argument 1 would be ill-formed:
gimbal.cpp:54:40: error: no matching function for call to ‘mavsdk::Gimbal::Gimbal(<brace-enclosed initializer list>)’
   54 |     auto gimbal = Gimbal{system.value()};
      |                                        ^
In file included from gimbal.cpp:10:
/usr/include/mavsdk/plugins/gimbal/gimbal.h:324:5: note: candidate: ‘mavsdk::Gimbal::Gimbal(const mavsdk::Gimbal&)’
  324 |     Gimbal(const Gimbal& other);
      |     ^~~~~~
/usr/include/mavsdk/plugins/gimbal/gimbal.h:324:5: note:   conversion of argument 1 would be ill-formed:
/usr/include/mavsdk/plugins/gimbal/gimbal.h:52:14: note: candidate: ‘mavsdk::Gimbal::Gimbal(std::shared_ptr<mavsdk::System>)’
   52 |     explicit Gimbal(std::shared_ptr<System> system); // new
      |              ^~~~~~
/usr/include/mavsdk/plugins/gimbal/gimbal.h:52:14: note:   conversion of argument 1 would be ill-formed:
/usr/include/mavsdk/plugins/gimbal/gimbal.h:39:14: note: candidate: ‘mavsdk::Gimbal::Gimbal(mavsdk::System&)’
   39 |     explicit Gimbal(System& system); // deprecated
      |              ^~~~~~
/usr/include/mavsdk/plugins/gimbal/gimbal.h:39:14: note:   conversion of argument 1 would be ill-formed:

julianoes, 2023-10-18:
Ok, how do you have mavsdk installed?

shivamydv2001, 2023-10-19:
I cloned github link.

julianoes, 2023-10-19:
And maybe installed from the .deb file?

shivamydv2001, 2023-10-20:
I think, I have done with that (.deb file )  . So can you provide any  documention link or something . I have given lot of time to solve

julianoes, 2023-10-20:
If you install it from deb, then you need to use the v1.4 branch, not main.

shivamydv2001, 2023-10-20:
**Thanks for your reply** @Julianoes

**Now I am getting this ..**...

**[ 50%] Building CXX object CMakeFiles/battery.dir/battery.cpp.o**
In file included from /home/shivam/MAVSDK/examples/battery/battery.cpp:7:
/usr/include/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough.h:9:10: fatal error: plugin_base.h: No such file or directory
    9 | #include "plugin_base.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/battery.dir/build.make:63: CMakeFiles/battery.dir/battery.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/battery.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

julianoes, 2023-10-20:
Please do a clean build and then provide all steps you are entering, otherwise I don't know what you're doing and can't reproduce it and can't help you!

shivamydv2001, 2023-10-20:
**I tried it** 

**Build and Try Example**
1.   git clone https://github.com/mavlink/MAVSDK.git --recursive
2.   cd MAVSDK
3.   cd examples
4.   cd takeoff_and_land/
5.   cmake -Bbuild -H.
6.   cmake --build build -j4

shivamydv2001, 2023-10-20:
if I am doing any mistake please tell me it would be appriciated

julianoes, 2023-10-20:
You need to use the v1.4 branch as I've written above.

```

    git clone https://github.com/mavlink/MAVSDK.git
    cd MAVSDK
    git switch v1.4
    git submodule update --init --recursive
    cd examples
    cd takeoff_and_land/
    cmake -Bbuild -H.
    cmake --build build -j4

```

reedev, 2023-10-20:
To me, it looks like there is an installed MAVSDK via a package or so, otherwise header files would not be used from /usr/include/mavsdk right?

reedev, 2023-10-20:
Ok, then I wonder why you would want to mix these and not go back to basic and only build from source code first, and make sure that's working first?

reedev, 2023-10-20:
ok, yes in the end when you are in control, you are right

shivamydv2001, 2023-10-20:
**Still I am getting this error** 
**Now I am using same command which you @julianoes sent to me**

In file included from /home/shivam/MAVSDK/examples/takeoff_and_land/takeoff_and_land.cpp:8:
/usr/include/mavsdk/plugins/action/action.h:16:10: fatal error: plugin_base.h: No such file or directory
   16 | #include "plugin_base.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/takeoff_and_land.dir/build.make:63: CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/takeoff_and_land.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

julianoes, 2023-10-20:
Which deb version do you have installed?

shivamydv2001, 2023-10-21:
mavsdk_0.37.0_ubuntu20.04_amd64.deb

shivamydv2001, 2023-10-21:
mavsdk_0.37.0_ubuntu20.04_amd64.deb

Is this MAVSDK library not compatible with this deb version

julianoes, 2023-10-21:
You should use the latest version v1.4.x

shivamydv2001, 2023-10-25:
Any other option with version 1.32 ?

closed 2023-10-30 by julianoes