mavlink/MAVSDK #2161: MavlinkPassthrough supporting for rgpc

david6096, 2023-10-20:
I want to process custom mavlink message from android platform,but MavlinkPassthrough plugin not exported to mavsdkserver through protobuf rgpc.any scheduling for this?

JonasVautherin, 2023-10-20:
It is not scheduled. @divyanshupundir is using https://github.com/divyanshupundir/mavlink-kotlin in parallel to MAVSDK-Java, I think.

If you only want to do raw mavlink you can look directly into mavlink-kotlin, I would say.

julianoes, 2024-05-20:
This would be nice and is something I have planned to do. However, it's not exactly my day job, so if you want to accelerate it, consider [throwing something in the hat](https://github.com/sponsors/julianoes).

julianoes, 2024-05-27:
Duplicate of #1459 .

closed 2024-05-27 by julianoes