mavlink/MAVSDK #2163: "Sending message failed" on main branch - Cannot Upload Missions on PX4 or ArduPilot Systems

OliverHeilmann, 2023-10-20:
I have a CI pipeline which builds mavsdk and runs some various sitl tests, checking that I can connect to hardware, send missions etc. My runs on Wednesday were working as expected and thereafter, I have been getting lots of issues with sending missions to PX4 and ArduPilot flight controllers (see below).

[image]

Has anyone encountered this? Hope there is a fix because it is quite a big failure mode to fail sending missions... For those wondering why I am not using a stable release version, there are some important fixes which have not been released yet and are in main currently.

Thanks for your help and let me know if you need more detail!

JonasVautherin, 2023-10-20:
> When I use the below commit version on Main, my code works as before so it looks like something in the last two pull requests has caused this issue.

Oh good catch, thanks for reporting this, that's helpful!

JonasVautherin, 2023-11-19:
What's the status here? Is that now fixed?

OliverHeilmann, 2023-12-02:
Sorry for the delay, just tested the latest main branch (debug) and the PX4 sitl environment works again i.e. mavlink messages are sending/ receiving correctly which is great.

I did also test the latest branch with ArduPilot sitl and got the following issue:
[image]

You can see that the `AUTO` mode change was getting rejected in flight. Happy for this issue to be closed and a new one opened for this new problem though!

julianoes, 2024-05-20:
There was no more follow up. I'm assuming everything is working.

closed 2024-05-20 by julianoes