mavlink/MAVSDK #2168: MAVSDK introduction page is out of date

stevenjowens, 2023-10-28:
https://mavsdk.mavlink.io/main/en/

On the Introduction page:

1) In the "Getting Started" section:

Check out the quickstart guide for [C++](https://mavsdk.mavlink.io/main/en/cpp/quickstart.html) and [Python](https://mavsdk.mavlink.io/main/en/python/quickstart.html). And no matter which language you are using, use the [C++ Guide](https://mavsdk.mavlink.io/main/en/cpp/guide.md) to learn how to perform common tasks and use the library in general.

The link to the C++ guide (https://mavsdk.mavlink.io/main/en/cpp/guide.md) gets  a 404 error.

This URL works: 

https://mavsdk.mavlink.io/main/en/cpp/guide/

2) In the "Getting Help" section:

It links to the now-defunct Slackchat, but not to the Discord.

julianoes, 2023-10-30:
Thanks for the notice.

If you feel like fixing it, here is the source: https://github.com/mavlink/MAVSDK-docs

stevenjowens, 2023-10-30:
I'll fix the C++ guide link.
Looks like somebody already fixed the slack/discord issue.

julianoes, 2024-05-20:
Here is one fix that looked missing: https://github.com/mavlink/MAVSDK-docs/pull/282