mavlink/MAVSDK #2172: How to enable Avoid ADSB mode?

qishen, 2023-11-07:
Hello, I have a use case for the [Avoid ADSB flight mode](https://ardupilot.org/copter/docs/flight-modes.html#flight-modes) in Ardupilot and it seems that's not in the default `FlightMode` in MAVSDK. I wonder what's the best practice to use an unsupported flight mode in MAVSDK and would you accept a PR to add a new plugin for Avoid ADSB mode?

julianoes, 2023-11-08:
The best way - but that's a bit a bigger task - would be to implement common flight modes for ArduPilot and MAVSDK.
Until then, could you add the flight mode here?
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/core/ardupilot_custom_mode.h

qishen, 2023-11-09:
The AvoidAdsb mode is already in `ardupilot_custom_mode.h` for both the copter and plane.
[https://github.com/mavlink/MAVSDK/blob/54f643c89b6b29a013a7d9fe8da2ebb411eb350b/src/mavsdk/core/ardupilot_custom_mode.h#L40](https://github.com/mavlink/MAVSDK/blob/54f643c89b6b29a013a7d9fe8da2ebb411eb350b/src/mavsdk/core/ardupilot_custom_mode.h#L40)

I know `FlightMode` is a set of common flight modes for both PX4 and Ardupilot while the AvoidAdsb mode is not there. We are still investigating AvoidAdsb mode and it seems that it's more like a feature to be turned on rather than a mode that you can switch to from a different mode according to the [Ardupilot ADSB documentation](https://ardupilot.org/copter/docs/common-ads-b-receiver.html). 

>Entry into this mode is automatic when avoidance is necessary based on the parameters below. Exit is also automatic when the threat has passed.

>In ArduPilot firmware versions 4.0 and later, the entry into this mode can be enabled or disabled via an RC channel switch by setting the channel’s RCx_OPTION = 38 (ADSB Avoidance En). If the RC PWM is >1800us, then entry into this mode is enabled if a threat presents.

But we still want a function or a new plugin that can be used to turn on the AvoidAdsb feature in MAVSDK. Do we still need to add it to the common `FlightMode` even though it's already in `ardupilot_custom_mode.h`?

qishen, 2023-11-09:
By the way, there is a typo about the mode name that it should be `AvoidAdsb` not `AvoidAdbs`.

julianoes, 2023-11-09:
If you want to fix the typo, just create a pull request, thanks.

julianoes, 2023-11-09:
And then we should work on a flight mode interface that works for all flight stacks and is String based.

julianoes, 2023-11-09:
If you want this to appear at the top of my todo list, consider [sponsoring me](http://github.com/julianoes/sponsors).

qishen, 2023-11-10:
Thank you for the quick reply! Appreciate your dedication to those projects so I just made a small donation from my personal fund. I will also let other team members know and see if they would like to sponsor you too.

What exactly is String-based? Do you mean an AvoidAdsb flight mode interface that works for both PX4 and Ardupilot when you say "all flight stacks"?

julianoes, 2023-11-10:
Yes, I mean just a generic interface like `set_mode(std::string mode_name)`, that will be compatible with the future common flight modes, which are also string based, at least the custom ones.

qishen, 2023-11-13:
Thanks @julianoes and I was thinking adding a new plugin for the unimplemented `AvoidAdsb` mode in the same way as `Offboard` in MAVSDK. Does it make sense?

qishen, 2023-11-13:
By the way, do you have any experience with the `AvoidAdsb` mode? We are still investigating how to enable the `AvoidAdsb` mode and according to the official documentation you can't just send a mavlink command to turn on `AvoidAdsb` mode like you do in switching to `AUTO` mode or `GUIDED` mode because the entry and exit of `AvoidAdsb` mode is automatically managed depending on the ADS-B signals received from nearby drones or in other words if the threats are still present.

julianoes, 2023-11-13:
So what else would this plugin do other than switching to the mode? Aha, so it's anyway automatic!

qishen, 2023-11-13:
So initially we want to experiment with the switching between `AvoidAdsb` mode and other modes as part of a research project. Recently we found out in the documentation that it's all automatic but we have trouble simulating the ADS-B signals from multiple drones in SITL or verify if the ADS-B signals are actually sent to the drone. It seems we can still set it to `AvoidAdsb` mode in MAVProxy with a command and we do not see the behavior of exiting the `AvoidAdsb` mode when we turn off the ADS-B signal. Appreciate if you have some knowledge about that part or direct us to the right person who knows. Thanks!

qishen, 2023-11-13:
Assuming the Ardupilot documentation is matched with the actual behavior of a drone, we only need two functions to turn on and off the `AvoidAdbs` feature in MAVSDK APIs. Note that I call it a feature rather than a mode because according to the documentation the mode switching is automatically managed but anyway we can't assume the documentation is always correct or the code implementation is correct. 

Hope you have time to add the interface in a feature branch and I will try to experiment with the implementation to make it work. The other day I was trying to make some changes in the [MAVSDK-Proto](https://github.com/mavlink/MAVSDK-Proto) to include a new plugin but failed to generate the code with some issues in building.

julianoes, 2023-11-13:
> two functions to turn on and off the AvoidAdbs feature 

Is is a command or a parameter?

qishen, 2023-11-13:
According to the documentation I quote below, it should be parameters but somehow you can still set it to `AvoidAdsb` with a mavlink command to change the mode while there is no guarantee that it will work correctly.

>ArduPilot includes a flight mode, AVOID_ADSB, that attempts to avoid manned vehicles based on the ADS-B sensor’s output. Entry into this mode is automatic when avoidance is necessary based on the parameters below. Exit is also automatic when the threat has passed.
To enable this feature connect with a Ground Station and set the following parameters:
[AVD_ENABLE](https://ardupilot.org/copter/docs/parameters.html#avd-enable): set to “1” to enable ADS-B based avoidance (param refresh may be necessary after setting this)
[AVD_F_DIST_XY](https://ardupilot.org/copter/docs/parameters.html#avd-f-dist-xy): the horizontal distance in meters that should be considered a near-miss
[AVD_F_DIST_Z](https://ardupilot.org/copter/docs/parameters.html#avd-f-dist-z): the vertical distance in meters above or below the vehicle that should be considered a near-miss
[AVD_F_TIME](https://ardupilot.org/copter/docs/parameters.html#avd-f-time): how many seconds in advance of a projected near-miss (based on the vehicle’s current position and velocity) the vehicle should begin the AVD_F_ACTION.
[AVD_F_ACTION](https://ardupilot.org/copter/docs/parameters.html#avd-f-action): controls how the vehicle should respond to a projected near-miss (i.e. 2:Climb Or Descend, 3:Move Horizontally, 4:Move Perpendicularly in 3D, 5:RTL or 6:Hover)
[AVD_F_RCVRY](https://ardupilot.org/copter/docs/parameters.html#avd-f-rcvry): sets how the vehicle will behave after the vehicle has cleared the near-miss area (i.e. 1 = resume previous flight mode)
Note: there are equivalent “Warn” parameters (i.e. AVD_W_DIST_XY) that can be used to adjust when warnings to the pilot will appear on the ground station.
In ArduPilot firmware versions 4.0 and later, the entry into this mode can be enabled or disabled via an RC channel switch by setting the channel’s RCx_OPTION = 38 (ADSB Avoidance En). If the RC PWM is >1800us, then entry into this mode is enabled if a threat presents.

julianoes, 2023-11-14:
If it is parameters, you should be able to make it work using the Param plugin.

qishen, 2023-11-14:
Thanks and I think the Param plugin works except that one param setting `RCx_OPTION = 38` failed after 5 retries. We still have some troubles running a SITL simulation with fake ADS-B signals so I wonder if MAVSDK supports reading the ADS-B signals received by the drone and download it to the ground control? Is Transponder plugin the right plugin to explore?

julianoes, 2023-11-14:
You can try Transponder, yes. I don't know what the state of it is though.

closed 2026-09-14 by julianoes