mavlink/MAVSDK #2173: How to use FTP Server and Client working together

fraviofii, 2023-11-09:
I'm trying to test the execution of FTP Server and Client of Mavlink and I am having some difficulties.

I cannot say that it is not working, because maybe I do not know how to use it.

The approach that I am using now is the following:

* Launch the PX4 simulator: https://github.com/jonasvautherin/px4-gazebo-headless
* On a terminal screen, launched the FTP server using the following command: `./ftp_server udp://:14540 /tmp
* On another terminal screen, launched the FTP client using the following command: `./ftp_client udp://:14550 190 dir

Is that the correct approach?

The `ftp_server`  indicates that it is working. Here's the output:

```
# ./ftp_server udp://:14540 /tmp
[07:35:16|Info ] MAVSDK version: v1.4.0-565-gb5453327 (mavsdk_impl.cpp:24)
Waiting for system
[07:35:16|Info ] New system on: 172.17.0.2:14580 (with system ID: 1) (udp_connection.cpp:199)
[07:35:16|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:714)
[07:35:16|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[07:35:17|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:222)
[07:35:17|Debug] Discovered 1 component(s) (system_impl.cpp:554)
[07:35:17|Debug] MAVLink: info: GCS connection regained	 (system_impl.cpp:248)
Starting FTP server
Mavlink FTP server running.

Remote:       udp://:14540:/tmp
Component ID: 190
```

The `ftp_client` does not seem to find the server. Here's the output:

```
# ./ftp_client udp://:14550 190 dir
[08:06:45|Info ] MAVSDK version: v1.4.0-565-gb5453327 (mavsdk_impl.cpp:24)
[08:06:45|Info ] New system on: 172.17.0.2:18570 (with system ID: 1) (udp_connection.cpp:199)
[08:06:45|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:714)
[08:06:45|Debug] Component Autopilot (1) added. (system_impl.cpp:379)
[08:06:45|Debug] New system ID: 245 Comp ID: 190 (mavsdk_impl.cpp:714)
[08:06:45|Debug] Component Ground station (190) added. (system_impl.cpp:379)
[08:06:45|Debug] Discovered 1 component(s) (system_impl.cpp:554)
[08:06:45|Debug] Discovered 1 component(s) (system_impl.cpp:554)
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault (core dumped)
```

I understand that this segmentation fault should not happen, but it is not clear for me if the my test is correct.

Could you please give me some advice on how to proceed?

julianoes, 2023-11-09:
This is based on the latest `main` where I have updated the FTP stuff (#2060)?

If so, then thanks for finding the first bug.

To help with debugging, just run it with `gdb --args`, then type `run`, hit enter. When it crashes, just type `backtrace` and copy it here.

Thanks!

julianoes, 2024-05-20:
No follow up. Given this has been changed a bunch I'm closing this.

closed 2024-05-20 by julianoes