mavlink/MAVSDK #2178: Reading additional messages

ygratz, 2023-11-14:
Hi,
I need to get the following rpt's:
HIGH_LATENCY2 (235)
EKF_STATUS_RPT (193)
POWER_STATUS (125)

What do I need to do in order to receive these messages? and to read their content?
Thanks.

julianoes, 2023-11-14:
If you use C++ you can use MavlinkPassthrough to access these messages. Otherwise, you'd have to make a pull request for the telemetry module to add the data that you require.

ygratz, 2023-11-15:
Hi,
I followed your advise and it works with MavlinkPassthrough.
I do have a problem with: mavlink_msg_ekf_status_report.h. This h file is missing.

When looking at: https://github.com/mavlink/c_library_v2/ most of the messages are at /common.
The mavlink_msg_ekf_status_report.h is at /ardupilotmega.
It seems MAVSDK does not include it.

What would be the most correct way to add this library as well? is there a way to configure MAVSDK to include it? 
Thanks

JonasVautherin, 2023-11-15:
It should work if you build MAVSDK with /ardupilotmega, given that [it does include /common](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L3) (that MAVSDK needs).

In the MAVSDK superbuild, when building mavlink, there is a [`-DMAVLINK_DIALECT`](https://github.com/mavlink/MAVSDK/blob/main/third_party/mavlink/CMakeLists.txt#L9) which is [set by the superbuild](https://github.com/mavlink/MAVSDK/blob/main/CMakeLists.txt#L8).

So it should work if you make a clean build of MAVSDK, and in the configure step (the first one, which has something like `-Bbuild/default` in it), add `-DMAVLINK_DIALECT=ardupilotmega`.