mavlink/MAVSDK #2183: MAVSDK remote hangs

MIKE9708, 2023-11-15:
Hi, i was trying to to use the MAVSDK python library to communicate with a drone in a  simulation ( the simulation is based on the PX4 repository using gazebo has a simulation tool ) hosted on a different machine. When i run the simulation and try :

``
drone = System()
await drone.connect(system_address="udp://ip_addr_of_target_machine:14540")
``

the script remains hanged, while on the simulation logs im able to see:

`INFO [mavlink] partner IP: ip_addr_of_MAVSDK_machine
`

Now looking on the PX4 [documentation](https://docs.px4.io/v1.12/en/simulation/) i read that there are a series of option in order to achieve what i want to do , i tried  **Enable UDP Broadcasting** , **Enable Streaming to Specific Address**,and   ** Use MAVLink Router**, without being able to get any result.

Can anyone help me understand  if i am doing something wrong or the PX4 docu solution should be modified?
Thanks

julianoes, 2023-11-15:
You have it backwards. You can't connect to UDP 14540 being published on another computer.

What you can do is:
`drone = System() await drone.connect(system_address="udp://:14540")`

And now enable broadcast, or specifically set the IP in PX4 `mavlink start ... -t IP`.

MIKE9708, 2023-11-15:
Actually the UDP port that i am using is 14550 on the mavsdk side i followed [this](https://docs.px4.io/v1.12/en/simulation/#use-mavlink-routerl)

JonasVautherin, 2023-11-15:
And does this work then?

```python
drone = System()
await drone.connect(system_address="udp://:14540")
```

MIKE9708, 2023-11-15:
Yes in localhost it works

JonasVautherin, 2023-11-15:
And your problem is when the simulator is running on another machine? Do you have mavlink-router running on that other machine and sending MAVLink to the machine that is running MAVSDK?

MIKE9708, 2023-11-15:
yeah i got mavlink router running on the machine with the STL and from the other machine i am using mavsdk for comunicating

MIKE9708, 2023-11-15:
I've also read [this issue](https://github.com/mavlink/MAVSDK/issues/2149), where the guy was using the port 18570 on mavsdk side

JonasVautherin, 2023-11-15:
Can you show your mavlink-router configuration file?

MIKE9708, 2023-11-15:
This my config file:

```

[UdpEndpoint QGC]
Mode = Normal
Address = 192.168.124.1 # MAVSDK machine
Port = 14550

[UdpEndpoint SIM]
Mode = Eavesdropping
Address = 127.0.0.1 # SITL
Port = 14550

```

JonasVautherin, 2023-11-16:
Seems like it should work with `udp://:14550` on the MAVSDK machine... :thinking:

MIKE9708, 2023-11-16:
So in theory my config file is correct right?

JonasVautherin, 2023-11-16:
Seems like it. Does QGC receive MAVLink if you run it instead of MAVSDK on your MAVSDK machine?

MIKE9708, 2023-11-16:
Ok but QGControl on the SITL machine or the mavsdk machine, because if i launch it on the SITL machine it is able to detect the drone

JonasVautherin, 2023-11-16:
And on the MAVSDK machine?

MIKE9708, 2023-11-16:
Yes Qgroundcontrol works and is able to detect the other machine

MIKE9708, 2023-11-16:
Ok i tried with another machine and it seems to work fine, i don't really know what was the problem

closed 2023-11-16 by MIKE9708