mavlink/MAVSDK #2187: MavlinkPassthrough  message handles are not unique

c-devine, 2023-11-24:
When subscribing to MavlinkPassthrough messages, the returned MessageHandle is not unique.  This causes problems when unsubscribing.  The handle's _id does increment when subscribing a second time to the same message, but starts at 1 for each message id.

An example is shown below.  If you unsubscribe to handle2, both message subscriptions are unsubscribed (they both have _id = 1). 

```
auto passthrough = MavlinkPassthrough{system.value()};

auto handle1 = passthrough.subscribe_message(MAVLINK_MSG_ID_HOME_POSITION,[](mavlink_message_t message) {
        std::cout << "message1 : " << message.msgid << std::endl;
    });

auto handle2 = passthrough.subscribe_message(MAVLINK_MSG_ID_HEARTBEAT,[](mavlink_message_t message) {
        std::cout << "message2 : " << message.msgid << std::endl;
    });

passthrough.unsubscribe_message(handle2);
```

julianoes, 2023-11-25:
Oh, that doesn't look right. Thanks for reporting.

The subscription happens here: 
https://github.com/mavlink/MAVSDK/blob/9a6416b5981d46d08c3ecd2332f460156628e6e0/src/mavsdk/plugins/mavlink_passthrough/mavlink_passthrough_impl.cpp#L200-L205
then:
https://github.com/mavlink/MAVSDK/blob/9a6416b5981d46d08c3ecd2332f460156628e6e0/src/mavsdk/core/system_impl.cpp#L99-L103

If I understand this right, we need to also pass the message ID when unsubscribing. It's not pretty but it would solve the problem, right?

c-devine, 2023-11-26:
Yeah, that looks like it would work.  Sending in the message ID would let you get the right CallbackList (if it exists) to unsubscribe from, instead of calling unsubscribe on every CallbackList.

julianoes, 2023-11-27:
Do you feel like making a pull request to fix it? :wink:

c-devine, 2023-12-02:
Just submitted a PR.  The update will require folks to refactor if they were using the old unsubscribe method.

julianoes, 2024-05-20:
Fixed in #2189.

closed 2024-05-20 by julianoes